robotology/yarp #596: Wrong link to the newest Windows version?

pauljurczak, 2015-10-05:
The main page link to Windows binaries is: [ http://sourceforge.net/projects/yarp0/files/yarp2/]( http://sourceforge.net/projects/yarp0/files/yarp2/), where the most current version is 2.3.63. The newest Windows binaries v2.3.64 seem to be at [https://github.com/robotology/yarp/releases](https://github.com/robotology/yarp/releases), instead.

pattacini, 2015-10-05:
Thanks @pauljurczak for pointing it out.

We have been migrating packages from SourceForge to GitHub, thus it might be that some links are not up-to-date yet.

Could you please tell us which link in particular is still old?
I've checked those contained here http://wiki.icub.org/wiki/ICub_Software_Installation, and at first sight they seem fine.

pauljurczak, 2015-10-05:
@pattacini The link on your main github page [https://github.com/robotology/yarp](https://github.com/robotology/yarp) to Windows binaries is old. I suspected it is due to repository migration.

commit 10e90c8d (2015-10-05): fix #596 thanks @pauljurczak
closed 2015-10-05 by pattacini