robotology/yarp #598: Compiling error with YARP_USE_GTK2 on OsX

iron76, 2015-10-06:
I am getting the following error. 

```
In file included from /Users/iron/Code/yarp/src/yarpscope-gtk/src/Debug.cpp:9:
In file included from /Users/iron/Code/yarp/src/yarpscope-gtk/src/Debug.h:15:
/usr/local/include/glibmm-2.4/glibmm/ustring.h:267:13: error: expected ';' at end of declaration list
  ~ustring() noexcept;
```

The erro can be removed by setting `YARP_USE_GTK2` to `OFF`.

drdanz, 2015-10-06:
Recent gtkmm versions require c++11 enabled.
Can you try recompiling using `cmake -DCMAKE_CXX_FLAGS=-std=c++11`?

iron76, 2015-10-06:
Then I get another error :laughing: 

```
[omissis]/yarp/src/libYARP_OS/src/BottleImpl.cpp:645:28: error: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wreserved-user-defined-literal]
    ACE_OS::sprintf(buf,"%"YARP_INT64_FMT,x);
```

drdanz, 2015-10-07:
Can you update and try again? I just fixed this and a few other issues with c++11 enabled...
Also instead of changing `CMAKE_CXX_FLAGS` you  can try enabling the `YARP_EXPERIMENTAL_CXX1` option

iron76, 2015-10-08:
New error:
```
[omissis]/yarp/src/libYARP_sig/src/IplImage.cpp:258:14: error: case value
      evaluates to 2147483656, which cannot be narrowed to type 'int'
      [-Wc++11-narrowing]
        case IPL_DEPTH_8S:
```

drdanz, 2015-10-08:
Ok, this one should also be fixed now... what's next? :stuck_out_tongue_closed_eyes:
Perhaps it would be faster if we do it together...

iron76, 2015-10-09:
No need to do it together because it finally works :+1: Thanks!

drdanz, 2015-10-09:
Cool :sunglasses:

I'm not closing this yet, because we should have a check on the gtk/sigc++ version and eventually enable c++11 only when required and only for the guis that actually require it.

traversaro, 2015-10-17:
New remainder : revert robotology/codyco-superbuild@66ffa56 when this bug is fixed .

DanielePucci, 2015-10-29:
Hi guys,
I've just joined the group of OsX friends, and it seems to me that I have I similar problem, probably the same. The output of the compilation is the following:
[image]

DanielePucci, 2015-10-29:
Sorry guys, I did not enable the flag `YARP_EXPERIMENTAL_CXX1`. It works now!

closed 2015-11-02 by drdanz

drdanz, 2015-11-02:
@DanielePucci @traversaro @iron76 @francesco-romano enabling `YARP_EXPERIMENTAL_CXX11` and/or c++11 should no longer be required, please reopen if for some reason it is still broken.

commit df015b01 (2015-11-02): Import updated modules from cmake-3.5 and YCM and fix GTK2 c++11 issues

All GTK related modules now use imported targets. This means that for
CMake 3.1 or later, c++11 is automatically enabled. for older versions,
it is manually enabled wherever it is required.

NOTE: conf/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake initial
comment changed from new style to old style.

Fixes #598
CC-Issue: robotology/icub-main#232