robotology/yarp #601: Broken link to pdf manual

pauljurczak, 2015-10-07:
The link to [official documentation in pdf](http://wiki.icub.org/yarp/specs/dox/min/latex/refman.pdf) on [yarp wiki page](http://wiki.icub.org/yarp/) is broken.

pauljurczak, 2015-10-07:
*People* link is broken too.

pattacini, 2015-10-07:
Thanks for reporting.
@mbrunettini could you please have a look into this?

mbrunettini, 2015-10-08:
It seems that the PDF creation is missing in the doxygen makefile.
@drdanz : can you enable it again?

traversaro, 2015-10-08:
That page ( http://wiki.icub.org/yarp/ ) has quite a lot of outdated links (bugs on sourceforge, source on sourceforge). Perhaps it make sense to just delete it and redirect it to http://wiki.icub.org/yarpdoc/ ?

lornat75, 2015-10-08:
Indeed, @mbrunettini can you get access to the page and redirect it?

mbrunettini, 2015-10-08:
Page redirected.

closed 2015-10-08 by mbrunettini