robotology/yarp #603: GUIs / Visual Modules not opening when launched from yarpmanager

rlober, 2015-10-09:
When I try to launch GUIs any visual modules from `yarpmanager` they fail to open. When I launch them from the terminal they work fine. Why might this be? I pulled/compiled yesterday.

```
[ERR] (Calib_Cameras) cannot run yarpview on icub-desktop : cannot run yarpview on /icub-desktop
[ERR] (Calib_Cameras) cannot run yarpview on icub-desktop : cannot run yarpview on /icub-desktop
[ERR] (Calib_Cameras) cannot run frameGrabberGui2 on icub-desktop : cannot run frameGrabberGui2 on /icub-desktop
[ERR] (Calib_Cameras) cannot run frameGrabberGui2 on icub-desktop : cannot run frameGrabberGui2 on /icub-desktop
[ERR] (Skin_Gui_All) cannot run skinManagerGui on icub-desktop : cannot run skinManagerGui on /icub-desktop
[ERR] (Skin_Gui_All) cannot run iCubSkinGui on icub-desktop : cannot run iCubSkinGui on /icub-desktop
[ERR] (Skin_Gui_All) cannot run iCubSkinGui on icub-desktop : cannot run iCubSkinGui on /icub-desktop
[ERR] (Skin_Gui_All) cannot run iCubSkinGui on icub-desktop : cannot run iCubSkinGui on /icub-desktop
[ERR] (Skin_Gui_All) cannot run iCubSkinGui on icub-desktop : cannot run iCubSkinGui on /icub-desktop
[ERR] (Skin_Gui_All) cannot run iCubSkinGui on icub-desktop : cannot run iCubSkinGui on /icub-desktop
[ERR] (Skin_Gui_All) cannot run iCubSkinGui on icub-desktop : cannot run iCubSkinGui on /icub-desktop
[ERR] (Skin_Gui_All) cannot run iCubSkinGui on icub-desktop : cannot run iCubSkinGui on /icub-desktop
```

https://github.com/robotology/yarp/issues/595 May be related but I am only like 10% sure on that.

Thanks!

alecive, 2015-10-09:
Do you have the `icub-desktop` node up and running?

rlober, 2015-10-09:
Yeah

alecive, 2015-10-12:
If they work, it might be that you did not compile them (either because you flagged the `CREATE_GUIS` to `OFF` manually, or because you didn't install `Qt` [or you didn' install it properly]).

rlober, 2015-10-12:
Hey @alecive 

The GUIs work when I launch them directly from a terminal, but not when I launch them from `yarpmanager`. Everything is compiled correctly. 

Is there any way to get more verbose output from `yarpmanager` than: 
`cannot run ... on icub-desktop ...`?

alecive, 2015-10-12:
@rlober sorry I forgot this little important detail. @apaikan any suggestion?

alecive, 2015-10-22:
@rlober are you able to launch the same GUIs through an `ssh`? I.E. go to another machine, `ssh` into `icub-desktop`, and then run `iCubSkinGui` (or any other gui for that matters) from the same console. It might be a display issue.

rlober, 2015-10-29:
@alecive 

I am in the process of completely overhauling our clusters and I think I will solve this problem in the process. I will go ahead and close the issue but if it comes up again after the overhaul then I will let you know.

closed 2015-10-29 by rlober