robotology/yarp #610: envelope mismatch

arrenglover, 2015-10-15:
When writing on, and reading from, a yarp::os::Port at high rates (up to 1Khz) envelope mismatch can occur between the data packet and the envelope. Reading from the port is called as such:

> yarp::os::port p;
> yarp::os::Bottle dummy;
> p.read(dummy);

and then

> yarp::os::Stamp st;
> p.getEnvelope(st);

We believe that between the call to read the data and the call to read the envelope, the internal thread of the yarp::os::Port is performing another read and updating to the next envelope. When the call to p.getEnvelope is made the envelope retrieved is the secondary envelope, not the envelope corresponding to the retrieved data. The problem only occurs at high communication rates.

We believe possible solutions include:
1. an option to perform both a data read and envelope read in the same function.
2. the option to lock the underlying reading thread until both data and envelope is read.

@drdanz , @apaikan

pattacini, 2015-10-15:
@arrenglover 

##### Caveat
Reading data by means of normal `Port` at high rate is conceptually wrong.
Switch to `BufferedPort<Bottle>` instead.

drdanz, 2015-10-15:
@pattacini I keep hearing this, but can you explain me why is that conceptually wrong? If you need blocking read() and write() at a high rate, why shouldn't you use the `Port`? Where is this documented?

pattacini, 2015-10-15:
Simply because blocking operations at very high communication rate (`~ 1 ms`) cannot be sustained by the infrastructure considering also the calling threads.

drdanz, 2015-10-15:
So it's just unsustainable by the infrastructure, and therefore you will not reach the desired communication rate? Anyway, even though it cannot be achieved right now with the current infrastructure, it does not mean that one day we will not have an infrastructure able to support such a rate... And what if both processes run on the same machine, perhaps using shared memory as carrier? Perhaps `1 ms` can be sustained in this case.
In my opinion "using `Port`" is not conceptually wrong and, if you don't want to drop data, `BufferedPort` will not help if you use **strict** `read()` and `write()` calls. The issue is eventually that you cannot achieve the desired communication rate, with the current infrastructure, without dropping packets. Am I wrong?

Anyway I *think* (but I should check how `BufferedPort` works internally) that the race condition could happen also on the `BufferedPort` between `read()` and `getEnvelope()` the envelope could be updated by the reading thread. One solution if you want to be sure to read the right envelope, could be to use the `BufferedPort` with a callback. In that case, if I remember correctly, the envelope should not be updated until the callback returns.

About the "Port" and this particular issue, If it is correct that the envelope received is the one from the _next_ packet, I think a relatively easy fix could be to change it to save the envelope of the last read somewhere, and have `getEnvelope()` always return the envelope saved after the last `read()` call. @lornat75 @apaikan what do you think?

pattacini, 2015-10-15:
My comment has nothing to do with the **envelope mismatch**, hence `BufferedPort` can be bugged too in this respect. I was only warned by the rate `1 KHz` that came alongside with the operation `Port::read()` :smirk: 

"_One day_" is a bit too far in the future and now, unfortunately, it's very unlikely to sustain a `1 KHz` rate without dropping packets, unless the sender / receiver threads do almost "_nothing_".

If the scenario is the one described above `1 KHz` + `Port::read()` I would be a bit surprised that the only problem is in the envelope mismatch (which is actually a bug), meaning that I would expect significant ripples in the threads rate (to be considered as an application issue).

arrenglover, 2015-10-16:
@pattacini

Thanks for the advice, rest assured we are using buffered ports throughout the project. However I was noticing 'dropped bottles' even when using strict write/read and tcp connections. The standard yarp::os::Port was then just used for testing to make the connection as simple as possible. The 'dropped bottles' where found to be not dropped at all, but just the read of the envelope mismatched with the data.

pattacini, 2015-10-16:
@arrenglover 

Good to know `BufferedPort` are used :smile: 

The **`strict`** attribute of a `BufferedPort` only makes sure that the sender waits until it can shoots over the network, meaning that its previous message has been sent out (but there's no guarantee it has been received), whereas a standard sender `Port` will be waiting until the receiver is done with its `read()` at the cost of uncontrolled and unpredictable fluctuation of the thread rates.

It might be a subtle difference but actually it turns to be crucial at those high rates and it's likely it does explain the different behaviors in terms of "_dropped bottles_"

barbalberto, 2015-10-16:
@arrenglover 

I made some test, and it indded happens when too much time passes between the read and the getEnvelope so that another message is received. I simulated it by putting some artifical delay in between the two.

If the network is overflowed by traffic bad thing could always happen, but if the priority in you case is to have data and envelope syncronized you can use the following solution:

``` code
/** Create a class to handle the callback after receiving a message from the buffered port.  */
class  StreamingCallback : public yarp::os::BufferedPort<yarp::os::Bottle>
{
protected:
    Stamp lastStamp;   // this is my envelope

public:
    StreamingCallback() { };

    /** Callback function.
     *  @param bottle is the data being received.
     */
    virtual void onRead(yarp::os::Bottle& bottle)
    {
        // When I get here, it means I've received a message, it is like doing a call to read()
        yarp::os::Time::delay(0.5);   // waste some time
        getEnvelope(lastStamp);      // get the envelope
        std::cout << "Received bottle is " << bottle.toString() << "; stamp is " << lastStamp.getTime() <<  std::endl;
       // in this way envelope is in synch with data because both are read inside the callback. If other messages are received, they will wait in the queue for the next callback to be called (or dropped)
    }
};

int main(int argc, char **argv)
{
    Network yarp;
//     BufferedPort<yarp::os::Bottle>    readPort;
    StreamingCallback    readPort;
    Stamp     time;

    readPort.open("/reading");
    readPort.useCallback();
    yarp::os::Network::connect("/writing", "/reading");

    while(1)
    {
        // dummy main code
         yarp::os::Time::delay(1);
    }

    return 0;
}
```

The advantage of using the buffered port in this case is that you can customize the callback and do all your opereation inside. This will assure the read and get envelope get serialized before reading a new data. Of course if you are not fast enough consuming the data you'll lose some messages.

I tested it with the following producer, where both the content of the message   and the envelope are set to the same progressive number:

``` code
int main(int argc, char **argv)
{
    Network yarp;
    BufferedPort<yarp::os::Bottle>    writePort;
    Stamp     time;
    int progressiveNumber = 0;

    writePort.open("/writing");

    while(1)
    {
        Bottle    &data = writePort.prepare();
        data.clear();
        data.addInt(progressiveNumber);
        time.update(progressiveNumber);
        writePort.setEnvelope(time);
        writePort.write();

        yarp::os::Time::delay(0.01);
        progressiveNumber++;
    }

    return 0;
}
```

The resullt is the following
````
Received bottle is 2354; stamp is 2354
Received bottle is 2403; stamp is 2403
Received bottle is 2453; stamp is 2453
Received bottle is 2502; stamp is 2502
Received bottle is 2551; stamp is 2551
Received bottle is 2600; stamp is 2600
Received bottle is 2650; stamp is 2650
Received bottle is 2699; stamp is 2699
Received bottle is 2748; stamp is 2748
Received bottle is 2798; stamp is 2798
Received bottle is 2847; stamp is 2847
Received bottle is 2896; stamp is 2896
Received bottle is 2945; stamp is 2945
Received bottle is 2995; stamp is 2995
Received bottle is 3044; stamp is 3044
Received bottle is 3093; stamp is 3093
```

You see like 50 dropped messages between 2 following read due to the timing of the sender and the receiver, but the envelope is always in synch.

arrenglover, 2015-10-16:
@barbalberto 

I can confirm that the envelope and bottle appear to be in synch using the bufferedPort in combination with the callback functionality (which is exactly the way most of our modules are set-up :) ). As such I also observe no bottle loss using a strict read and tcp connection. Perfect.

The bug still exists for someone who has a valid reason for using ports and does not read the envelope immediately after the data, e.g. if the thread is interrupted between the two reads.

lornat75, 2015-10-17:
Good to hear that callback in buffered port fixes this issue, however I agree with @drdanz that it would be preferable to store the timestamp during the call to read and return it later. the current implementation may easily cause a lot of problems.

The discussion about BufferedPort versus Port is a little off-topic... there is no intrinsic gain in a BufferedPort, by default it decouples timing bw sender and receivers, with a Port you have to be more careful to get the same behavior. Maybe the documentation could be improved to clarify this issue which is quite confusing.

pattacini, 2015-10-18:
@lornat75 yeah that was off-topic, but anyway the choice between **`Port & BufferedPort`** is always worth discussing since it has quite a big impact on the intercomm design and as you said it's prone to confuse people.

A `Port` might be turned into something similar to a `BufferedPort` (there must be a way) and of course a `BufferedPort` can be used in blocking read() but I was referring to them more as two _pointers_ to two different paradigms: **coupled vs. decoupled** communication.