robotology/yarp #611: [Windows] Cmake does not accept 64b Yarp, works with 32b

WurmD, 2015-10-19:
In Win8
MSVS 2013
Cmake v3.1.0-rc3 (only 32b is available)
find_package(YARP REQUIRED) does not accept 64b version of Yarp (neither v2.3.63 or v2.3.64, both installed from binaries)

```
CMake Error at CMakeLists.txt:11 (find_package):
  Could not find a configuration file for package "YARP" that is compatible
  with requested version "".

  The following configuration files were considered but not accepted:

    C:/Program Files/robotology/yarp-2.3.64/cmake/YARPConfig.cmake, version: 2.3.64 (64bit)
```

drdanz, 2015-10-19:
CMake 32 bit is able to create projects for 64 bit. You are probably using the 32 bit of the generator. See [[1]](https://cmake.org/cmake/help/v3.1/manual/cmake-generators.7.html) and [[2]](https://cmake.org/cmake/help/v3.1/generator/Visual%20Studio%2012%202013.html)

WurmD, 2015-10-19:
That is correct!
Sorry for the bother :)

drdanz, 2015-10-19:
No problem.
Closing

closed 2015-10-19 by drdanz