robotology/yarp #612: ControlBoardWrapper has a limit for MAX_JOINTS_ON_DEVICE

liesrock, 2015-10-20:
ControlBoardWrapper.h on master branch, has an enum called MAX_VALUES_FOR_ALLOCATION_TABLE_TMP_DATA, with a field MAX_JOINTS_ON_DEVICE set to 32: for WALK-MAN robot(33 joints) this value is too low when we use a single wrapper for the whole-robot.

lornat75, 2015-10-20:
We can certainly increase the number or make it dynamic.

I suggest that you do not use a single chain, an interface with 32 joints is very annoying to use. Can you quantify the overhead splitting into separate legs, arms and torso+head?

barbalberto, 2015-10-20:
Unfortunatly it is required for interaction with ROS. The topic has to iclude all joints of the robot to work reliably. We are facing the same issue with the CER simulator in Gazebo/rViz

randaz81, 2015-10-20:
I remember a ros node that performs the merge from multiple joint state publishers. Isn't this solution more practical?

liesrock, 2015-10-20:
We needed to use a single chain (only one wrapper) most of all because of CPU usage and YARP overhead related with many open wrappers(head, arms, legs, torso, hands): I will quantify this overhead in the next messages (I have to collect the old data).

barbalberto, 2015-10-21:
'Practical' or ' annoying' depends on a lot of things, mainly the taste of the people J
Policy can change depending on the actual case. For walkman, threads are running at 1ms while in the iCub they run at 10 ms. This means 10 times more cpu usage and this was really an issue for on-board PC. Adding another node just for merging topic means other cpu usage and a delay which is not negligible when running at 1ms.
Furthermore all their project is about whole robot interaction and all their algorithm are meant to use most of the robot, if not all, so split it means more work without benefits.

My opinion is that some best practice can be advised but forcing the user to write the code on a specific way only because of some hardcoded value is synthom of poor programming. This is something we were  aware since the beginning ... time has come for a fix.

randaz81, 2015-10-21:
@barbalberto yeah, my comment was related to: `We are facing the same issue with the CER simulator in Gazebo/rViz` which does not run at 1KHz.
However, i'm interested in @liesrock data about the overhead.

barbalberto, 2015-10-21:
Ah, ok! Sorry ... I was too focused on walkman

liesrock, 2015-10-22:
About the overhead data: 
when we used the robotInterface with 7 motion control wrappers(head, arms, legs, torso, hands) with 1ms or 2ms (we tried both) thread period + 4 FT wrappers, two at 1ms and two at 10ms, we had a CPU usage of 120% without any modules using the robotInterface. With the walking module (YARP RateThread running at 1ms/2ms) we had a CPU usage of 160% for the robotInterface. Inside the walking RateThread we quite often observed spikes of 20ms respect to the requested thread period (1ms/2ms).

When we started using only 1 motion control wrapper(with 33 joints) with 2ms thread period and 1 FT wrapper with 10ms thread period, we had a CPU usage of 15% without any modules using the robotInterface. With the walking module (YARP RateThread running at 3ms) we had a CPU usage of 25% for the robotInterface. Inside the walking RateThread we observed spikes of 2ms respect to the requested thread period (3ms) .

I completely agree that is annoying use a single interface with 33 joints, but it is our only solution to make the robot walk using the YARP robotInterface: with the former described overhead(120% to 160%), it is basically impossible.

lornat75, 2015-10-26:
I understand. Maybe you can consider a tradeoff, I guess head and hands (wris ?) are not controlled with the other joints bc they do not have a large impact on stability and walking. These parts are probably controlled by separate modules and it makes sense to separate them. And (maybe) for these parts the rate can be reduced to 10ms.

drdanz, 2015-10-26:
For the CER robot in gazebo, the problem is the `MAX_DEVICES=5` hardcoded limit. I suggest to address this issue as quickly as possible.

barbalberto, 2015-10-27:
It is in the #619  pull request, I just want to do some test to be sure the remapping is correct

barbalberto, 2015-11-02:
Merged. Closing issue

closed 2015-11-02 by barbalberto