robotology/yarp #614: yarpmotorgui should be able to reconnect

drdanz, 2015-10-21:
Using gazebo, whenever the model is removed, the yarpmotorgui disconnects and must be closed and re-opened. This is really annoying, it would be very useful if it could use persistent connection, or perhaps if it had a button reconnect.

alecive, 2015-10-21:
+1000 :+1:

iron76, 2015-10-23:
:-1: Personally I don't like persistent connections. It makes sense to me that a connection is forever lost if something was connected to a module that was then destroyed.

alecive, 2015-10-23:
Me neither, but the feature is nice. This is why we were rather thinking of a reconnect button (or automatic reconnection) :smile:

lornat75, 2015-10-23:
+1 for the reconnect button.

barbalberto, 2015-10-23:
Please remove the noisy message "connection has been lost, you should save and restart" ... I hate it!!
:imp:

alecive, 2015-10-23:
It will be gone for good :smile: