robotology/yarp #616: ITorqueControl adding multi-joint PIDs method

francesco-romano, 2015-10-21:
We need the following two methods to be implemented on the eth based robots:
```
virtual bool setTorquePids(const int n_joint, const int *joints, const Pid *pids)
virtual bool getTorquePids(const int n_joint, const int *joints, Pid *pids)
```

@barbalberto any idea on how to do this?
I tried to download the icub_firmware_shared, but I think I won't be able to compile it...

traversaro, 2015-10-21:
@francesco-romano `icub_firmware_shared` does not need to be compiled, it just a collection of headers.

barbalberto, 2015-10-22:
No PID interface has the group method by design right now, neither position. If we add it it would make sense to do it for all of the interfaces.

Is the intended usage to change them at "high rate"? If this is the case, maybe the right place is the streaming port instead of rpc like the other PID functions.

P.S.  icub_firmware_shared in not the  right place to look at
P.S.2 actually icub_firmware_shared contains code that  should be compiled, but it doesn't ... iCub repo is compiling

randaz81, 2015-10-22:
Rejected: No streaming. Mandatory only rpc. No high refresh rate.

barbalberto, 2015-10-22:
@lorenat what do you think, is it ok to add this group function for pids? For all pids?

francesco-romano, 2015-10-22:
Just to specify: this request was on adding the group functions in the interface, not about their streaming/rpc implementation.

After discussion with @randaz81 and @barbalberto we decided that this feature can be useful, but as a matter of fact, given the kind of use we'd like to do on the robot it is of very low priority.

Will be solved/close in the future

lornat75, 2015-10-22:
Maybe streaming is not a good idea because setting pids is something that need to be reliable. Anyway I see the issue has been settled already.