robotology/yarp #617: Wrong RPATH for Plugins

drdanz, 2015-10-22:
Plugins are not installed in lib, but in lib/yarp, therefore the RPATH should be different.

drdanz, 2015-10-22:
@barbalberto Can you please confirm that this is fixed?

barbalberto, 2015-11-02:
Issue is fixed, repos that wants to use plugins have to add those lines in the cmakelists

```
add_install_rpath_support(LIB_DIRS "${CMAKE_INSTALL_FULL_LIBDIR}"                                   # Libraries
                          BIN_DIRS "${CMAKE_INSTALL_FULL_BINDIR}"                                   # Binaries
                                   "${CMAKE_INSTALL_PREFIX}/${{REPO_NAME}_DYNAMIC_PLUGINS_INSTALL_DIR}"     # Plugins
                          USE_LINK_PATH)
```

closed 2015-11-02 by barbalberto

drdanz, 2015-11-02:
@barbalberto Awesome! Should we update the documentation for creating plugins somewhere?

barbalberto, 2015-11-02:
yes, I guess the right page is this one http://www.yarp.it/add_a_plugin.html
I'll do it soon

commit 66619270 (2015-10-22): Use AddInstallRPATHSupport from YCM to handle RPATH

Also fix RPATH issues with plugins

Fixes #617