robotology/yarp #618: yarpdataplayer streams halved images

gsaponaro, 2015-10-22:
Hi, I am trying to use some old datadumps (of Image type), but when I stream the images to a viewer, I can only see the top half:

[image]

**yarpdataplayer** (Qt version) is giving this warning:
````
your specific IPL format (BGR depth 8 -> bgr) does not match your YARP format ()
Making a copy instead of just wrapping...
````

Steps to reproduce the problem:
* with **yarpdataplayer**, play [this datadump](http://users.isr.ist.utl.pt/~gsaponaro/temp/dataset_for_halved_image_issue.zip) (7MB)
* start a **yarpview**
* connect the image stream to the viewer

Interestingly, the obsolete **yarpdataplayer-gtk** streams the images fine.

This could be related to https://github.com/robotology/yarp/commit/d268a75019985fb520d3fe92648de1fe9876adba.

vtikha, 2015-10-22:
@gsaponaro 
I have just tried the dataset on a MacOSX and a Windows machine with the latest yarp and I have a full image in the stream. The frame rate is low but looking at the timestamps it seems expected. Furthermore I do not have the BGR warning reported. Do you have yarp updated? The **the obsolete yarpdataplayer-gtk** should not be used as it no longer supported. **yarpdataplayer** is back complatible with old datadumps. 

[image]

gsaponaro, 2015-10-22:
I see. I tried it on two Linux machines with up-to-date yarp (Ubuntu 14.04) and I obtained the problem on both of them. I wonder if other Linux users can reproduce this problem, or the cause is elsewhere on my systems. :)

vtikha, 2015-10-22:
@gsaponaro I do not think this has to do with the operating system. Are you using yarpdataplayer with opencv or with yarp image? I have just committed a change that informs the user upon playing which library it is using for reading and sending the images.

gsaponaro, 2015-10-22:
OK: ````[INFO]USING OPENCV FOR SENDING THE IMAGES````.

Versions 2.4.9 (from source) and 2.4.8 (from Ubuntu packages) respectively.

gsaponaro, 2016-01-14:
Hi @vtikha, would it be possible to further analyze this issue? As I mentioned in the previous messages, my **yarpdataplayer** is relying on **OpenCV**. Ideally I'd like to use my old dumps for some POETICON++ development (it is faster for me than using the real robot, which at the moment is also under repair / unavailable).

vtikha, 2016-03-14:
commit 9a7d38a has fixed the halved images sent by ```yarpdataplayer``` on ubuntu 14.04

closed 2016-03-14 by vtikha

gsaponaro, 2016-03-14:
Cool, confirmed working here, thanks.