robotology/yarp #621: OS::PortTest failing in CI travis?

lornat75, 2015-10-23:
I noticed that in some PR travis CI fail with the follow error:

29/63 Test #29: OS::PortTest .......................***Exception: SegFault  3.77 sec
test 30
      Start 30: OS::PortReaderBufferTest

This problem seems to be unrelated with the code changed in PR and is sporadic, maybe there is a race condition.

@drdanz

traversaro, 2016-03-04:
Another related failed test: https://travis-ci.org/robotology/yarp/jobs/113625737 .

traversaro, 2016-04-19:
Probably related to https://github.com/robotology/WB-Toolbox/issues/11 .

traversaro, 2016-04-23:
All occurrences of this bug (Travis, https://github.com/robotology/WB-Toolbox/issues/11) appear to be on OS X .

traversaro, 2016-04-23:
Never mind: apparently I was able to replicate the issue on Ubuntu 16.04 , with the following command:
`ctest -R PortTest::Valgrind::MemCheck  --repeat-until-fail 10000 --output-on-failure` , with following final output: 
~~~
yarp(4069cc0): Thread being deleted
yarp(4069cc0): Thread being deleted
yarp: 0 | PortTest: checking interrupt for a port with pending reply...
yarp(4069cc0): register contact /out
yarp(4069cc0): opening for address tcp://localhost:10003
yarp(b63e700): Thread starting up
yarp(4069cc0): Child thread initializing
yarp(4069cc0): Child thread initialized ok
yarp: Port /out active at tcp://localhost:10003
yarp(4069cc0): register contact /in
yarp(4069cc0): opening for address tcp://localhost:10002
yarp(4069cc0): Child thread initializing
yarp(a23b700): Thread starting up
yarp(4069cc0): Child thread initialized ok
yarp: Port /in active at tcp://localhost:10002
yarp(4069cc0): working on connection /out to /in (connect)
yarp(4069cc0): query name /out
yarp(4069cc0): Configuration file: /home/pegua/.config/yarp/yarp_namespace.conf
yarp(4069cc0): query name /in
yarp(4069cc0): Configuration file: /home/pegua/.config/yarp/yarp_namespace.conf
yarp(4069cc0): ** asking tcp://out: [add] "tcp://in"
yarp(4069cc0): scanning network interface 127.0.0.1
yarp(4069cc0): scanning network interface 192.168.1.9
yarp(4069cc0): scanning network interface 192.168.122.1
yarp(4069cc0): scanning network interface ::1
yarp(4069cc0): scanning network interface fe80::7218:8bff:fedd:bfa7%3
yarp(b63e700): /out: PortCore received something
yarp(b63e700): new input connection to /out starting
yarp(b63e700): Child thread initializing
yarp(aa3c700): Thread starting up
yarp(b63e700): Child thread initialized ok
yarp(b63e700): new input connection to /out started ok
yarp(b63e700): /out: PortCore spun off a connection
yarp(b63e700): /out: / routine check of connections to this port begins
yarp(b63e700): /out: | checking connection ->-> 
yarp(b63e700): /out: \ routine check of connections to this port ends
yarp(aa3c700): TextCarrier::expectSenderSpecifier
yarp(aa3c700): Receiving input from admin to /out using text_ack
yarp(aa3c700): /out: asked to add output to tcp://in
yarp(aa3c700): query name /in
yarp(aa3c700): Configuration file: /home/pegua/.config/yarp/yarp_namespace.conf
yarp(aa3c700): /out: asked to remove connection /out->*->/in
yarp(aa3c700): scanning network interface 127.0.0.1
yarp(aa3c700): scanning network interface 192.168.1.9
yarp(aa3c700): scanning network interface 192.168.122.1
yarp(aa3c700): scanning network interface ::1
yarp(aa3c700): scanning network interface fe80::7218:8bff:fedd:bfa7%3
yarp(a23b700): /in: PortCore received something
yarp(a23b700): new input connection to /in starting
yarp(a23b700): Child thread initializing
yarp(be3f700): Thread starting up
yarp(a23b700): Child thread initialized ok
yarp(a23b700): new input connection to /in started ok
yarp(a23b700): /in: PortCore spun off a connection
yarp(a23b700): /in: / routine check of connections to this port begins
yarp(a23b700): /in: | checking connection ->-> 
yarp(a23b700): /in: \ routine check of connections to this port ends
yarp(be3f700): Port /in starting up, flushing routes /out->*->/in
yarp(be3f700): /in: asked to remove connection /out->*->/in
yarp: Receiving input from /out to /in using tcp
yarp(be3f700): expecting an index
yarp(be3f700): ConnectionState::expectIndex for /out->tcp->/in
yarp: Sending output from /out to /in using tcp
yarp(aa3c700): /out: / routine check of connections to this port begins
yarp(aa3c700): /out: | checking connection admin->text_ack->/out 
yarp(aa3c700): /out: | checking connection /out->tcp->/in 
yarp(aa3c700): /out: \ routine check of connections to this port ends
[... 208 lines cut ...]
==16265==    by 0x581795B: ??? (in /usr/lib/x86_64-linux-gnu/libsqlite3.so.0.8.6)
==16265==    by 0x581AEA6: ??? (in /usr/lib/x86_64-linux-gnu/libsqlite3.so.0.8.6)
==16265==    by 0x581B97B: ??? (in /usr/lib/x86_64-linux-gnu/libsqlite3.so.0.8.6)
==16265==    by 0x58519C4: ??? (in /usr/lib/x86_64-linux-gnu/libsqlite3.so.0.8.6)
==16265==    by 0x5851E3A: ??? (in /usr/lib/x86_64-linux-gnu/libsqlite3.so.0.8.6)
==16265==    by 0x5852928: ??? (in /usr/lib/x86_64-linux-gnu/libsqlite3.so.0.8.6)
==16265==    by 0x5854D8F: ??? (in /usr/lib/x86_64-linux-gnu/libsqlite3.so.0.8.6)
==16265==    by 0x58591E1: ??? (in /usr/lib/x86_64-linux-gnu/libsqlite3.so.0.8.6)
==16265==    by 0x5874FF9: ??? (in /usr/lib/x86_64-linux-gnu/libsqlite3.so.0.8.6)
==16265==    by 0x587D1A6: sqlite3_step (in /usr/lib/x86_64-linux-gnu/libsqlite3.so.0.8.6)
==16265== 
==16265== LEAK SUMMARY:
==16265==    definitely lost: 0 bytes in 0 blocks
==16265==    indirectly lost: 0 bytes in 0 blocks
==16265==      possibly lost: 23,064 bytes in 22 blocks
==16265==    still reachable: 2,081,979 bytes in 10,179 blocks
==16265==                       of which reachable via heuristic:
==16265==                         length64           : 26,776 bytes in 266 blocks
==16265==         suppressed: 0 bytes in 0 blocks
==16265== Reachable blocks (those to which a pointer was found) are not shown.
==16265== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==16265== 
==16265== For counts of detected and suppressed errors, rerun with: -v
==16265== ERROR SUMMARY: 11 errors from 11 contexts (suppressed: 1 from 1)
~~~

Tobias-Fischer, 2016-08-08:
I managed to get a dump, this might help for debugging. My guess is that `sending` in `PortCoreOutputUnit` should be a `mutex` (and probably `running` as well). Maybe @drdanz wants to fix another test ;-).

```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007fae6ba0f751 in yarp::os::impl::PortCoreOutputUnit::sendHelper (this=this@entry=0x7fae60004370) at /home/icub/robot/yarp/src/libYARP_OS/src/PortCoreOutputUnit.cpp:333
333	    if(replied && op->getSender().modifiesReply()) {
[Current thread is 1 (Thread 0x7fae69405700 (LWP 7801))]

Thread 5 (Thread 0x7fae5bfff700 (LWP 7797)):
#0  0x00007fae6c5f97ad in accept () at ../sysdeps/unix/syscall-template.S:84
#1  0x00007fae6c926511 in ACE_SOCK_Acceptor::accept(ACE_SOCK_Stream&, ACE_Addr*, ACE_Time_Value*, bool, bool) const () from /usr/lib/libACE-6.3.3.so
#2  0x00007fae6ba68742 in yarp::os::impl::SocketTwoWayStream::open (this=this@entry=0x7fae54001fb0, acceptor=...) at /home/icub/robot/yarp/src/libYARP_OS/src/SocketTwoWayStream.cpp:64
#3  0x00007fae6ba6dbd5 in yarp::os::impl::TcpFace::read (this=0x234ac40) at /home/icub/robot/yarp/src/libYARP_OS/src/TcpFace.cpp:87
#4  0x00007fae6b9f8f4d in yarp::os::impl::PortCore::run (this=0x2347ff0) at /home/icub/robot/yarp/src/libYARP_OS/src/PortCore.cpp:184
#5  0x00007fae6ba707b5 in theExecutiveBranch (args=0x2347ff0) at /home/icub/robot/yarp/src/libYARP_OS/src/ThreadImpl.cpp:96
#6  0x00007fae6c903f56 in ACE_OS_Thread_Adapter::invoke() () from /usr/lib/libACE-6.3.3.so
#7  0x00007fae6c5f06fa in start_thread (arg=0x7fae5bfff700) at pthread_create.c:333
#8  0x00007fae6b092b5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

Thread 4 (Thread 0x7fae69c06700 (LWP 7796)):
#0  0x00007fae6c5f97ad in accept () at ../sysdeps/unix/syscall-template.S:84
#1  0x00007fae6c926511 in ACE_SOCK_Acceptor::accept(ACE_SOCK_Stream&, ACE_Addr*, ACE_Time_Value*, bool, bool) const () from /usr/lib/libACE-6.3.3.so
#2  0x00007fae6ba68742 in yarp::os::impl::SocketTwoWayStream::open (this=this@entry=0x7fae5c002950, acceptor=...) at /home/icub/robot/yarp/src/libYARP_OS/src/SocketTwoWayStream.cpp:64
#3  0x00007fae6ba6dbd5 in yarp::os::impl::TcpFace::read (this=0x22c8aa0) at /home/icub/robot/yarp/src/libYARP_OS/src/TcpFace.cpp:87
#4  0x00007fae6b9f8f4d in yarp::os::impl::PortCore::run (this=0x22c7f00) at /home/icub/robot/yarp/src/libYARP_OS/src/PortCore.cpp:184
#5  0x00007fae6ba707b5 in theExecutiveBranch (args=0x22c7f00) at /home/icub/robot/yarp/src/libYARP_OS/src/ThreadImpl.cpp:96
#6  0x00007fae6c903f56 in ACE_OS_Thread_Adapter::invoke() () from /usr/lib/libACE-6.3.3.so
#7  0x00007fae6c5f06fa in start_thread (arg=0x7fae69c06700) at pthread_create.c:333
#8  0x00007fae6b092b5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

Thread 3 (LWP 7800):
#0  0x00007fae6c5f07be in __exit_thread () at ../sysdeps/unix/sysv/linux/exit-thread.h:36
#1  start_thread (arg=0x7fae68c04700) at pthread_create.c:478
#2  0x00007fae6b092b5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

Thread 2 (Thread 0x7fae6cd6f740 (LWP 7350)):
#0  yy_reduce (yyruleno=<optimised out>, yypParser=<optimised out>) at sqlite3.c:133175
#1  sqlite3Parser (yyp=0x234b818, yymajor=55, yyminor=..., pParse=0x234b568) at sqlite3.c:3186
#2  0x00007fae6bf87aa4 in sqlite3RunParser (pParse=0x234b568, zSql=0x22c9918 "SELECT id, ns, name, value FROM tags WHERE rid = 188 AND ns IS NULL AND name = 'host'", pzErrMsg=0x7ffe416253b0)
    at sqlite3.c:135186
#3  0x00007fae6bf89c26 in sqlite3Prepare (db=0x221ef98, zSql=0x22c9918 "SELECT id, ns, name, value FROM tags WHERE rid = 188 AND ns IS NULL AND name = 'host'", nBytes=-1, saveSqlFlag=1, 
    pReprepare=<optimised out>, ppStmt=0x7ffe416254a8, pzTail=0x0) at sqlite3.c:112596
#4  0x00007fae6bf89d3c in sqlite3LockAndPrepare (db=0x221ef98, zSql=0x22c9918 "SELECT id, ns, name, value FROM tags WHERE rid = 188 AND ns IS NULL AND name = 'host'", nBytes=-1, saveSqlFlag=1, pOld=0x0, 
    ppStmt=0x7ffe416254a8, pzTail=0x0) at sqlite3.c:112688
#5  0x00007fae6bf8a008 in sqlite3_prepare_v2 (db=<optimised out>, zSql=<optimised out>, zSql@entry=0x22c9918 "SELECT id, ns, name, value FROM tags WHERE rid = 188 AND ns IS NULL AND name = 'host'", 
    nBytes=<optimised out>, nBytes@entry=-1, ppStmt=<optimised out>, ppStmt@entry=0x7ffe416254a8, pzTail=<optimised out>, pzTail@entry=0x0) at sqlite3.c:112764
#6  0x00000000004d78e2 in SqliteTripleSource::query[abi:cxx11](Triple&, TripleContext*) (this=0x2232590, ti=..., context=<optimised out>)
    at /home/icub/robot/yarp/src/libYARP_serversql/src/SqliteTripleSource.h:141
#7  0x00000000004d911c in NameServiceOnTriples::query (this=this@entry=0x221eac0, portName=..., act=..., prefix=..., nested=nested@entry=false)
    at /home/icub/robot/yarp/src/libYARP_serversql/src/NameServiceOnTriples.cpp:50
#8  0x00000000004da2b9 in NameServiceOnTriples::cmdQuery (this=this@entry=0x221eac0, act=..., nested=nested@entry=false) at /home/icub/robot/yarp/src/libYARP_serversql/src/NameServiceOnTriples.cpp:133
#9  0x00000000004ddf87 in NameServiceOnTriples::apply (this=0x221eac0, cmd=..., reply=..., event=..., remote=...) at /home/icub/robot/yarp/src/libYARP_serversql/src/NameServiceOnTriples.cpp:627
#10 0x00000000004d3573 in yarp::name::NameServerConnectionHandler::apply (this=0x7ffe41626330, reader=..., writer=0x2349ac0, lock=<optimised out>)
    at /home/icub/robot/yarp/src/libYARP_name/include/yarp/name/NameServerConnectionHandler.h:58
[... cut ...]
#5  0x000000000048a0ee in ServiceUser::run (this=0x7ffe41628640) at /home/icub/robot/yarp/tests/libYARP_OS/PortTest.cpp:200
#6  0x00007fae6ba707b5 in theExecutiveBranch (args=0x23465a0) at /home/icub/robot/yarp/src/libYARP_OS/src/ThreadImpl.cpp:96
#7  0x00007fae6c903f56 in ACE_OS_Thread_Adapter::invoke() () from /usr/lib/libACE-6.3.3.so
#8  0x00007fae6c5f06fa in start_thread (arg=0x7fae69405700) at pthread_create.c:333
#9  0x00007fae6b092b5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

drdanz, 2016-08-09:
@Tobias-Fischer Thanks for the dump, that's quite useful! I'll have a look at it as soon as I have some time.
How did you reproduce the bug? I haven't been able to reproduce it locally...

Tobias-Fischer, 2016-08-09:
Hi, I used the following snippet: `while /home/icub/robot/yarp/build/bin/harness_os "regression" "PortTest"; do :; done`. It might take a while though before it crashes.

Tobias-Fischer, 2016-08-11:
Hi, I think it's much easier than that. 
The protection `if (op!=NULL)` does not go all the way to the line where the segfault happens (`if(replied && op->getSender().modifiesReply())`), thus I think in the segfault case `op==NULL` and thus `op->getSender()` leads to a segfault. Thus, moving the closing `}` from line 325 down just before the `return` statement might fix the problem. 

I'm running the test now, if it doesn't fail by the end of the day, I'll create a PR.

drdanz, 2016-08-11:
@Tobias-Fischer Awesome! I'm looking forward for your patch :+1:

Tobias-Fischer, 2016-08-16:
This still happens :( https://travis-ci.org/robotology/yarp/jobs/152354037

PR #866 (2016-08-12): Fixes #621: Add check for NULL to fix potential segfault
This fixes #621. The addition of `op!=NULL` is the actual fix; moving `if(done) { ... }` up in the other protected area is just clean-up (and does not change the logic). 

I've written a small comment in the code, as just moving the `if(replied ...)` up in the other protected area is not enough. The call to `closeBasic` might set `op=NULL`, and thus there is another check required whether `op!=NULL`.

I had this running for ~16 hours without crash.

Would be great if you could merge @drdanz.
<a href='#crh-start'></a><a href='#crh-data-%7 [... cut]
commit 5060d456 (2016-08-12): Merge pull request #866 from robotology/fix/621

Fixes #621: Add check for NULL to fix potential segfault
commit ae07debd (2016-08-12): Fixes #621: Add check for NULL to fix potential segfault