robotology/yarp #623: mjpeg crash

randaz81, 2015-10-26:
module camCalibWithPose crashes with the following log (segfault, broken pipe).
Suspect bug in mjpeg carrier.

(gdb) bt full
#0  0x00007ffff5fcd88d in write () at ../sysdeps/unix/syscall-template.S:81
No locals.
#1  0x00007ffff5041d09 in ACE::send_n_i(int, void const*, unsigned long, unsigned long*) () from /usr/lib/libACE-6.0.3.so
No symbol table info available.
#2  0x00007ffff71d70ed in yarp::os::impl::SocketTwoWayStream::write(yarp::os::Bytes const&) () from /usr/local/src/robot/yarp/build-ubuntu/lib/libYARP_OS.so.1
No symbol table info available.
#3  0x00007ffff029add1 in send_net_data(unsigned char*, int, void*) () from /usr/local/src/robot/yarp/build-ubuntu/lib/yarp/yarp_mjpeg.so
No symbol table info available.
#4  0x00007ffff004220c in jpeg_finish_compress () from /usr/lib/x86_64-linux-gnu/libjpeg.so.8
No symbol table info available.
#5  0x00007ffff029b0b3 in yarp::os::MjpegCarrier::write(yarp::os::ConnectionState&, yarp::os::SizedWriter&) ()
   from /usr/local/src/robot/yarp/build-ubuntu/lib/yarp/yarp_mjpeg.so
No symbol table info available.
#6  0x00007ffff7253d9e in yarp::os::impl::Protocol::write(yarp::os::SizedWriter&) () from /usr/local/src/robot/yarp/build-ubuntu/lib/libYARP_OS.so.1
No symbol table info available.
#7  0x00007ffff7238209 in yarp::os::impl::PortCoreOutputUnit::sendHelper() () from /usr/local/src/robot/yarp/build-ubuntu/lib/libYARP_OS.so.1
No symbol table info available.
#8  0x00007ffff7239458 in yarp::os::impl::PortCoreOutputUnit::run() () from /usr/local/src/robot/yarp/build-ubuntu/lib/libYARP_OS.so.1
No symbol table info available.
#9  0x00007ffff72a05bc in theExecutiveBranch(void*) () from /usr/local/src/robot/yarp/build-ubuntu/lib/libYARP_OS.so.1
No symbol table info available.
#10 0x00007ffff5089b16 in ACE_OS_Thread_Adapter::invoke() () from /usr/lib/libACE-6.0.3.so
No symbol table info available.
#11 0x00007ffff5ccc182 in start_thread (arg=0x7fffc97fa700) at pthread_create.c:312
        __res = <optimized out>
        pd = 0x7fffc97fa700
        now = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140736573974272, 3729621361613693557, 1, 140736750173104, 140736573974976, 140736573974272, 
                -3729591678052244875, -3729634392865545611}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, 
              canceltype = 0}}}
        not_first_call = <optimized out>
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>
        __PRETTY_FUNCTION__ = "start_thread"
#12 0x00007ffff5fdc47d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
No locals.

drdanz, 2015-10-26:
The crash seems to be happening also using udp. Further testing to ensure that the crash is the same is required, though

drdanz, 2016-06-17:
Never seen this crash again. Closing

closed 2016-06-17 by drdanz