robotology/yarp #624: If YARP_CLOCK env variable is setted, yarp complains that is not initialized

traversaro, 2015-10-26:
~~~
Makas-Pro:iDynTree makaveli$ YARP_CLOCK=/clock yarp
yarp: YARP not initialized; create a yarp::os::Network object before using ports
yarp: Anonymous port active at tcp://192.168.36.36:60657
yarp: Receiving input from /clock to /192.168.36.36:60657 using tcp
This is the YARP network companion.
Call with the argument "help" to see a list of ways to use this program.
yarp: Removing input from /clock to /192.168.36.36:60657
~~~

This happened because the `NetworkClock` object is created before the `__yarp_is_initialized` variable is increased
(see https://github.com/robotology/yarp/blob/83466b733d89d2f3687542b047bcd5185d3f9299/src/libYARP_OS/src/Network.cpp#L622 ). 

This issue was introduced by https://github.com/robotology/yarp/pull/609 because before the actual creation of the `NetworkClock` was delay until the clock was actually used, typically long after the `NetworkBase::initMinimum()` method had returned. 

cc @barbalberto

barbalberto, 2015-10-28:
commit 3688021 should fix this issue

barbalberto, 2015-11-12:
This issue needs to be re-opened.

**Synthom**
RFModules hangs when network clock is used.

**Cause** 
Clock API are incoherent:
getTime and Time::now behave differently
isSystemClock check does not take into account ```network_clock_pending``` flag

**What happens**
RFModule::runModule() uses function  ```getTime``` to get the current time value which internally uses the ```isSystemClock()``` to detect whether the clock is a system, network or custom one. This funtion simply check if ```*clock``` is null and in this case return true.

When calling useNetworkClock the implementation set ```network_clock_pending = true;``` ***BUT*** does not create the clock, delaying the real creation of clock untill the first usage.
**Problem** The getTime() -> isSystemClock() does not count as 'first usage' because it ignore the network_clock_pending flag, therefore clock is considered system untill someone calls Time::now, Time::isValid or Time::isValid

As a consequence when the first Time::now is called, and therefore the network clock is created, RFModule hangs because it computes the 'time to sleep' using values coming from two different clock.
What it happens is "systemClock - networkClock" which result in "sleep for 40years".

The ```getClock``` right after calls to ```useXXXClock``` removed on https://github.com/robotology/yarp/commit/3688021090f652d79db2d8105eb2b625dff9a277
 was meant to prevent this error, but was causing clock to open unnamed port.

** Solution**
One possible solution is to make isSystemClock to take ```network_clock_pending``` into account by calling Time::isvalid before performing the check on the pointer, like Time::isValid does.

I'll test this solution and open a PR asap. This is for reminder/discussion

barbalberto, 2015-11-17:
Open a new issue for the new synthom

closed 2015-11-17 by barbalberto