robotology/yarp #628: yarpmotorgui should adjust the slider step according to max-min

pattacini, 2015-10-28:
Currently, the slider step established by the `yarpmotorgui` is fixed at `1 unit` irrespective of the actual joint span. We have to make it adjustable instead.

For example if the joint span is `[min,max]`, then the slider step could be `(max-min)/100.0` by default.
One further possibility could be also to let the user tune such slider step from within the gui.

This functionality is mandatory (and **urgently required** :smirk:) for those joints that have a very different spans from the usual ones, e.g. linear joints whose elongation is expressed in meters.

/cc @barbalberto @randaz81 @ale-git @alecive @vtikha @lornat75 @drdanz

lornat75, 2015-10-28:
I agree we could initially do 100 steps (esier).

pattacini, 2015-10-29:
[**Here**](http://doc.qt.io/qt-4.8/qslider.html) it is reported:

> The slider is the classic widget for controlling a bounded value. It lets the user move a slider handle along a horizontal or vertical groove and translates the handle's position into an **integer value** within the legal range.

Apparently **Qt** knows only integers, disregarding all the developments of mathematics in the field of real numbers... Is what I read really true? Isn't there already a slider class handling real numbers?

alecive, 2015-10-30:
We can scale everything from 0 to 100 and work with integers in the slider. Then, we can add a label that displays the real number instead. They can be decoupled I think.

pattacini, 2015-10-30:
I know that, but I'm still very disappointed that Qt does not have a native solution...

However, we **MUST** do things neatly, therefore we have to inherit a new slider class that wraps over **`QSlider`** and does the conversion internally. This way we can avoid messing up the code with conversions spread all around.

pattacini, 2015-10-30:
This [**SE post**](http://stackoverflow.com/questions/19003369/how-to-make-a-qslider-change-with-double-values) seems to provide some hints.

randaz81, 2015-11-03:
now implemented (e8e1100e83e00fa083429a174b097a6ad03444a8).
The behavior can be selected in the "view" menu of the gui.
Please test it carefully before merging to master.

randaz81, 2015-11-05:
It was found that the above fix for position slider is not enough.
The velocity slider whose range is fixed to (0-100) has to be modified as well. This requires a major change, since the admissibile velocity range for each joint has to be asked to robotInterface.

pattacini, 2016-01-29:
65e9cecbe543d9d40363c47a72cd9cc85690f71c fixes this issue.
Thanks @randaz81.
Closing.

closed 2016-01-29 by pattacini