robotology/yarp #629: yarpmanager: symbol lookup error

rlober, 2015-10-29:
Hello all,

I just did a completely fresh install of `yarp`, `icub-main` etc. and most everything seems to work except `yarpmanager`. When I try to launch it I get the following undefined symbol error:
```
ryan@ISIR2014057:~/Install/bin$ yarpmanager
yarpmanager: symbol lookup error: yarpmanager: undefined symbol: _ZTVN4yarp2os4impl20SystemInfoSerializerE
```
I demangled it through c++filt and got the following:
```
ryan@ISIR2014057:~/Install/bin$ c++filt _ZTVN4yarp2os4impl20SystemInfoSerializerE
vtable for yarp::os::impl::SystemInfoSerializer
```
Just FYI, yarpmanager-gtk works, and yarpmotorgui works.

Thanks in advance,
Ryan

drdanz, 2015-11-02:
@rlober This sounds like an issue in your system, otherwise the failure would have been on build, not on execution. Please check if you have more than one installation of YARP in your system.

rlober, 2015-11-04:
@drdanz 

I just upgraded from ubuntu 12.04 to 14.04 (yeah I know we are behind the times) and I am doing quite a bit of cleanup. My guess is something nasty is still lying around that is causing this. I will close this issue for now, but I can comment later when I resolve it.

closed 2015-11-04 by rlober

apaikan, 2015-11-05:
My guess is that you have some (old) Yarp libraries somewhere nesting in your system. 

* check which `yarpmanager` you are running: 
```
which yarpmanager
``` 

* check which libraries it is using:
```
ldd `which yarpmanager`
```