robotology/yarp #633: [MAC OS X] Can't launch applications through yarprun

jeljaik, 2015-11-05:
Hello there,
I have checked with other laptops to see if yarprun works fine and apparently it's a problem of my current system configuration. But I can't figure out what the problem is and I would appreciate any hint. The steps I follow are the following:
1. `yarpserver` 
2. `yarprun --server /jorh`
3. `yarprun --on /jorh --as whatever --cmd iCubGui`

These are my logs for each terminal after the final command.
From `yarpserver` I get
```bash
Jorhabibs-MacBook-Pro:~ jeljaik$ yarpserver
__   __ _    ____  ____  
\ \ / // \  |  _ \|  _ \ 
 \ V // _ \ | |_) | |_) |
  | |/ ___ \|  _ <|  __/ 
  |_/_/   \_\_| \_\_|    

Call with --help for information on available options
Using port database: :memory:
Using subscription database: :memory:
IP address: default
Port number: 10000
yarp: Port /root active at tcp://192.168.36.77:10000

Registering name server with itself:
 * register "/root" tcp "192.168.36.77" 10000
   + set "/root" ips "127.0.0.1" "192.168.36.77"
   + set "/root" process 758
 * register fallback mcast "224.2.1.1" 10000
   + set fallback ips "127.0.0.1" "192.168.36.77"
   + set fallback process 758
Name server can be browsed at http://192.168.36.77:10000/

Ok.  Ready!
 * register "/jorh"
   + set "/jorh" ips "127.0.0.1" "192.168.36.77"
   + set "/jorh" process 759
-> announce "/jorh"
 * register "..."
 * set "/tmp/port/1" ips "127.0.0.1" "192.168.36.77"
 * set "/tmp/port/1" process 761
-> announce "/tmp/port/1"
 * query "/tmp/port/1"
 * query "/jorh"
 * query "/jorh"
 * query "/tmp/port/1"
 * query "/jorh"
 * unregister "/tmp/port/1"
```

From `yarprun`
```bash
Jorhabibs-MacBook-Pro:~ jeljaik$ yarprun --server /jorh
^CJorhabibs-MacBook-Pro:~ jeljaik$ yarprun --server /jorh
STARTED: server=/jorh alias=whatever cmd=iCubGui pid=762
```

From the yarprun cmd:
```bash
Jorhabibs-MacBook-Pro:~ jeljaik$ yarprun --on /jorh --as whatever --cmd iCubGui
yarp: Port /tmp/port/1 active at tcp://192.168.36.77:10003
yarp: Sending output from /tmp/port/1 to /jorh using tcp
yarp: Removing output from /tmp/port/1 to /jorh
RESPONSE:
=========
762
STARTED: server=/jorh alias=whatever cmd=iCubGui pid=762
```

After this, I don't get `iCubGui` anywhere. I can launch it however without `yarprun` so I believe it's got nothing to do with its installation. Also when I try to launch an application through `yarpmanager` on a node different from `localhost` the modules won't start.

Any idea?

gsaponaro, 2015-11-05:
Did you install yarp and icub-main from binaries or from source?

Binaries -> no clue, hopefully some Mac OS X user can come along and help. :)

Source -> read on.

It could be that the commands launched via ```yarprun``` and ```yarpmanager``` do not see the correct environment variables like ```$PATH```, thus they cannot find binaries like ```iCubGui```.

Not sure about Mac OS X, but in Linux I would fix it like this:

First put the desired variables in your ```.bashrc_iCub``` file, example:
```
 export CODE=/home/yourname/code
 export YARP_ROOT=$CODE/yarp
 export YARP_DIR=$YARP_ROOT/build
 export ICUB_ROOT=$CODE/icub-main
 export ICUB_DIR=$CODE/icub-main/build
 export YARP_DATA_DIRS=$YARP_DIR/share/yarp:$ICUB_DIR/share/iCub
 export PATH=$PATH:$YARP_DIR/bin:$ICUB_DIR/bin
 export ICUB_ROBOTNAME=your_robotname
```

Then make sure that the rc file is loaded both during interactive (terminal) and non-interactive (yarprun) sessions, by putting these lines
```
# per-user environment variables (non-interactive and interactive mode)
source $HOME/.bashrc_iCub
```
at the beginning of ```/etc/bash.bashrc```, before the line that says ```[ -z "$PS1" ] && return```

See also:
* http://wiki.icub.org/wiki/Generic_iCub_machine_installation_instructions#icub_user_enviroment_variables (towards the end)
* http://wiki.icub.org/wiki/Debugging_problems_with_yarprun

iron76, 2015-11-06:
I am experiencing similar problems on my MacBook. @jeljaik any update on your side?

jeljaik, 2015-11-09:
Thank you @gsaponaro I installed both yarp and icub-main from sources, I have checked as you suggested the environmental variables seen by xterm through `xterm -hold -e "printenv"`but they seem to be updated correctly.  

[image]

francesco-romano, 2016-08-09:
I managed to replicate the issue.
The problem is if C++11 is enabled.
And the problem is occurring at line https://github.com/robotology/yarp/blob/master/src/libYARP_OS/src/ThreadImpl.cpp#L174 which with c++11 crashes with 
```
libc++abi.dylib: terminating with uncaught exception of type std::__1::system_error: thread::join failed: No such process
```

This is called by
https://github.com/robotology/yarp/blob/master/src/libYARP_OS/src/Run.cpp#L762

francesco-romano, 2016-08-11:
Good [stackoverflow](http://stackoverflow.com/questions/38848534/c11-thread-is-joinable-but-join-raises-an-exception?noredirect=1#comment65143061_38848534) gave me a hint:
> The problem was that a thread was created before fork() and after fork() there was no thread in app while C++ thought there's one. That's why join() on joinable thread threw an error ESRCH

PR #865 (2016-08-18): Fixes yarprun (at least on macOS)
Symptom  of the problem was the following uncaught exception:
```
libc++abi.dylib: terminating with uncaught exception of type std::__1::system_error: thread::join failed: No such process
```

Details of the problem:

> The problem is that the join was called on a thread object in a process different from the one which created it.
> This happened because of calls to `fork()`.
> 
> In particular we have to be careful when we call `fork()`: all the memory is duplicated but only one thread (the one calling `fork()`) survives.
> In general it can be very dangerous [... cut]
closed 2016-08-18 by drdanz