robotology/yarp #634: Streaming ports get stuck

barbalberto, 2015-11-05:
While using gazeControl module with saccade option on, the robot's head stops moving after few commands.
What has been observed is either one of the two phenomenon:
- iKinGaze stops sending command getting stuck inside the setPositions() 
- iCub's ControlBoardWrapper stops processing streaming commands and therefore gets overwhelmed by streaming messages creating a very long queue (more than 1000 messages). Even when the sender stops sending new commands, the queue is not consumed anymore.

Either process does not exit when closing, it has to be killed.

I found out that when removing the 'strict' option the ports does not get stuck anymore, but of course the behaviour is not correct.

From some preliminary test, My guess is that the strict option is buggy and causing a deadlock sometime in the write (GazeControl stopos sending commands) or in the read (huge queue of message is created).

This is probably related to #583

pattacini, 2015-11-09:
Is there any test to verify this functionality?

It seems that several times in the past (see https://github.com/robotology/icub-main/issues/129) a problem on the `writeStrict` along with the `position-direct` mode has been responsible for quite a big impact on our high-level software and was also difficult to be narrowed down.

barbalberto, 2015-11-09:
Not yet, I'll make one to use with the RTF asap

pattacini, 2015-11-09:
:+1: 
I bet this new test will help us considerably. Thanks!

lornat75, 2015-11-10:
It would be easier if you replicate the problem only using ports. In this case the test can go with the others in YARP.

lorejam, 2015-11-10:
Just to reinforce that in our case this happens ALWAYS and after just ONE command is sent on the /xd:i port.
We tried it several times.

pattacini, 2015-11-10:
@lorejam, yep the problem is deterministic: it does always happen, sometime on the sender side, other times on the receiver side.

pattacini, 2015-11-17:
Any progress on this issue?

pattacini, 2015-11-26:
@barbalberto could you deal with this tomorrow during **IFI Day 4**?

barbalberto, 2015-11-27:
I made a test and I'm not able to reproduce it. I and Ugo tried also on the real robot and the problem didn't show up anymore. @loreJam, do you still experience this issue?

pattacini, 2015-11-27:
Nasty heisenbug ...

pattacini, 2015-11-27:
For completeness, I think @barbalberto also tried to reproduce the problem with a pure port-based approach, being the result the same: apparently, everything works properly.

lorejam, 2015-11-27:
I can update YARP/iCub and test it...
...did you make relevant changes to YARP/iCub that could possibly have
solved the issue?
Otherwise, I bet the problem would still be there :)

-- 
===================================================================

Lorenzo Jamone - Research Associate
http://lorejam.blogspot.pt/
ljamone@isr.tecnico.ulisboa.pt <Tecnicojasv@isr.tecnico.ulisboa.pt>

VisLab - http://vislab.isr.ist.utl.pt
Instituto Superior Técnico - http://tecnico.ulisboa.pt/

Institute for Systems and Robotics, Torre Norte (7th floor).
Av. Rovisco Pais, 1
1049-001 Lisbon, Portugal

Office (room 7.25): +351218418299
Laboratory (room 7.26): +351218418050

====================================================================

lornat75, 2015-11-27:
it could be a strange interaction with the code in the remote control board...

pattacini, 2015-11-27:
@barbalberto's guess is that there might be a deadlock in one of the mutex of the port; however, no commits have been done recently in this respect to justify the changed behavior.

Our main aim now is to try to make it happen again, otherwise debugging will be difficult.

barbalberto, 2015-12-15:
This has been fixed by commit 2700fd4977b7247893a0df32ba73295a8c6ccebc
Bug was:  checkMotionDone took a mutex but never release it, so following request got stuck.

closed 2015-12-15 by barbalberto

pattacini, 2015-12-15:
Great! :tada:

lorejam, 2015-12-15:
Yee!
I will test as soon as I can and confirm ;)

Lore

lornat75, 2015-12-15:
Pretty serious bug, strange it did not cause deterministic, catastrophic effects... thanks for finding it!

lornat75, 2015-12-15:
...but this means that two subsequent checkMotionDone would have caused a deadlock, for example MotorTest should have triggered a deadlock: https://robotology.github.io/icub-tests/doxygen/doc/html/classMotorTest.html

barbalberto, 2015-12-15:
The point is that only group version was affected, and checkmotionDone group version is almost never used.

The mutex is shared among different functions, so this bug was *actually* creating a deadlock, just the blocked call was not the checkMotionDone but a different one using the same mutex. This made the bug a bit more subtle to spot.

Today me and @valegagge modified the test in order to use it, however a test dedicated to test only communication (all possible method) between wrapper and client would be nice to add and run before the testMotor.

lornat75, 2015-12-15:
if you have added a test that triggers this bug, and that now with the commit passes I think it is good enough.

gsaponaro, 2016-01-26:
We're a bit late to the party (this bug is already marked as solved), but we just wanted to report that this issue is solved on ```iCubLisboa01``` too: the gaze test with saccades described in https://github.com/robotology/icub-main/issues/238 works fine.

barbalberto, 2016-01-27:
Thanks for the feedback!