robotology/yarp #635: Euler to quaternions conversions

drdanz, 2015-11-06:
It would be useful to have methods to convert from euler angles (in their different forms) to quaternions, as many ros messages (i.e. /tf messages) use quaternions to represent the rotation.

barbalberto, 2015-11-06:
yae. for IMU ROS topic I made one here:
https://github.com/robotology/yarp/blob/master/src/libYARP_dev/src/modules/msgs/ros/include/yarpRosHelper.h

If it is usefule we can move in a more public place

traversaro, 2015-11-06:
## TL; DR: **Be extremely careful** :smile: 

### Long comment 

Consider that eulerAngles <--> quaternion conversions are quite tricky, especially near the euler angles singularity ( see [1], [2] or [3] just to mention the one that caused us many hours of debugging ). 

Relying on a widely used existing library (for iDynTree we used Eigen) is a way of reducing the risk of getting things wrong, but adding a new dependency on YARP is clearly not an option. 

Writing our own functions but with extensive unit testing (see [4]) comparing their result with existing implementation could also be a viable strategy. 
 
[1] : https://github.com/orocos/orocos_kinematics_dynamics/pull/32 
[2] :  https://github.com/ros/urdfdom_headers/issues/18 
[3] : https://github.com/robotology/idyntree/issues/79
[4] : https://github.com/ros/urdfdom/pull/67

pattacini, 2015-11-06:
@drdanz to keep the structure present in yarp I'd go for `quat2dcm` / `dcm2quat` rather than for `quat2euler` / `euler2quat`.

pattacini, 2015-11-27:
Unit test added too; still quite simple though, but better than nothing :smirk:

pattacini, 2015-11-27:
Closing for now.
Feel free to comment further on this.

closed 2015-11-27 by pattacini

barbalberto, 2015-11-27:
Can this be used to subsitute the mentioned case of IMU? https://github.com/robotology/yarp/blob/master/src/libYARP_dev/src/modules/msgs/ros/include/yarpRosHelper.h  convertEulerAngleYXZrads_to_quaternion()

pattacini, 2015-11-27:
You could do that relying on a pivot; something like

```cpp
Vector q=dcm2quat(rpy2dcm(angles));
```

traversaro, 2015-12-09:
@barbalberto I was checking the convention used in the different libraries for implementing quaternion support in iDynTree, and I just noticed that in YARP we adopted the scalar/vector representation (I guess for consistency for Matlab) [1], while ROS is using vector/scalar [2] . 
So YARP/ROS quaternion conversion should be handled with care. 

[1] : http://www.yarp.it/namespaceyarp_1_1math.html#abd6a7d78b72b3346987d6122c02ec534
[2] : http://docs.ros.org/api/geometry_msgs/html/msg/Quaternion.html

pattacini, 2015-12-09:
Before someone will rise the point that we should conform to `ROS` in this respect, I'd like to underline how `MATLAB` is a more appropriate baseline to stick to for math - of course - and that all the other methods we have in `Yarp` to deal with such transformations adhere to `MATLAB` convetion.
:smirk:

barbalberto, 2015-12-09:
I think the point here is that `serverInertial` opens a topic for talking with ROS, so that topic should send the data in the correct format.
So basically if we use the dcm function then the first value of dcm will became the last one for ROS, like:
```
ROS[0] = DCM[1]
ROS[1] = DCM[2]
ROS[2] = DCM[3]
ROS[3] = DCM[0]
```
right?

pattacini, 2015-12-09:
If you use:
```cpp
Vector q=dcm2quat(rpy2dcm(angles));
```
then:
```cpp
ROS[0] = q[1];
ROS[1] = q[2];
ROS[2] = q[3];
ROS[3] = q[0];
```

traversaro, 2015-12-09:
Actually my point was not completely correct: ROS never uses a "serialization" to R^4 for dealing with quaternion messages, it always uses the individual attributes `w`, `x`,`y`, `z` (that are unambiguous, at least from this point of view). 

The serialization is just used in our helper code in https://github.com/robotology/yarp/blob/master/src/libYARP_dev/src/modules/msgs/ros/include/yarpRosHelper.h , but then the "ROS" message is using the individual attributes: https://github.com/robotology/yarp/blob/f1c24cc7b2b53c7dc9f1227c6348f4d4f7aff57d/src/libYARP_dev/src/modules/ServerInertial/ServerInertial.cpp#L457 .

traversaro, 2015-12-09:
By the way, `scalar/vector` is also the one used in Eigen.

pattacini, 2015-12-09:
:+1: then!

traversaro, 2016-04-21:
Late comment on the RPY / Euler angles issue, I think this paper : https://www.astro.rug.nl/software/kapteyn/_downloads/attitude.pdf is a nice reference document to unambiguously define a choice of Euler angles.

traversaro, 2016-04-21:
CAVEAT: I just noticed that in this paper R represent the rotation matrix that applied to a vector expressed in world orientation it transforms it in body-fixed orientation (i.e. b_R_w ) while typically in robotics  R its transpose (i.e. w_R_b ).