robotology/yarp #636: [MAC OS X] Yarpscope not working properly from XML file.

jeljaik, 2015-11-09:
Hello guys,

I am trying to use `yarpscope` from XML without success, while using it with the `--remote` option works just fine. The following command:

`yarpscope --remote /icub/right_leg/state:o --index 4` 

works as expected while the following will constantly show zero. 

```bash
<?xml version="1.0" encoding="UTF-8" ?>
<portscope rows="1" columns="1" carrier="mcast">
    <plot gridx="0"
          gridy="0"
          hspan="1"
          vspan="1"
          title="Pitch (Blue)"
          size="60"
          minval="-5"
          maxval="5"
          bgcolor="LightSlateGrey">
        <graph remote="/icub/right_leg/state:o"
               index="4"
               color="Green"
               title="Pitch quatEKF"
               size="3"
               type="lines" />                       
    </plot>
</portscope>
```

jeljaik, 2015-11-09:
If the carrier is changed to `tcp` things work as expected. If `mcast` connections are not supported in the current version of `yarpscope` I guess this issue can be closed. Otherwise, it can be renamed to something more appropriate such as: 

**Yarpscope does not support mcast connections on Mac OS X**

vtikha, 2015-11-09:
@jeljaik True, but it seems that this is not a ``Yarpscope`` problem but is systemwide issue on Mac OS X. Same behaviour for the ``udp`` protocol.

jeljaik, 2015-11-09:
Oh ok, good to know :ok_hand:  Thanks

closed 2015-11-09 by jeljaik

iron76, 2015-11-09:
@vtikha I haven't seen any issue open. This is quite critical and I think it should be a priority. Should we open a proper issue?

vtikha, 2015-11-09:
@iron76 I remember that this came up a while ago and even talking with @paulfitz about it, but I have lost track of it. As you said, let's open a proper issue and since we now have a large amount of mac users we can test/debug it properly.