robotology/yarp #639: isSystemClock return true after useNetworkClock has been called

barbalberto, 2015-11-17:
Synthom
RFModules hangs when network clock is used.

Cause 
Clock API are incoherent:
getTime and Time::now behave differently
isSystemClock check does not take into account network_clock_pending flag

What happens
RFModule::runModule() uses function getTime to get the current time value which internally uses the isSystemClock() to detect whether the clock is a system, network or custom one. This funtion simply check if *clock is null and in this case return true.

When calling useNetworkClock the implementation set network_clock_pending = true; BUT does not create the clock, delaying the real creation of clock untill the first usage.
Problem The getTime() -> isSystemClock() does not count as 'first usage' because it ignore the network_clock_pending flag, therefore clock is considered system untill someone calls Time::now, Time::isValid or Time::isValid

As a consequence when the first Time::now is called, and therefore the network clock is created, RFModule hangs because it computes the 'time to sleep' using values coming from two different clock.
What it happens is "systemClock - networkClock" which result in "sleep for 40years".

The getClock right after calls to useXXXClock removed on 3688021
was meant to prevent this error, but was causing clock to open unnamed port.

** Solution**
One possible solution is to make isSystemClock to take network_clock_pending into account by calling Time::isvalid before performing the check on the pointer, like Time::isValid does.

I'll test this solution and open a PR asap. This is for reminder/discussion

traversaro, 2017-08-10:
This was fixed by https://github.com/robotology/yarp/pull/1277 I guess, cc @barbalberto .

barbalberto, 2017-08-10:
yes.
after calling `useNetworkClock()` the status is:
- `isSystemClock()` returns ` false` 
- `isNetworkClock()` returns `true`
- `Clock->isValid()`  will return `false` until the first time message has been received from the network, then it will return `true`.

closed 2017-08-10 by barbalberto