robotology/yarp #640: YARP tweak version, tags, and release cycle discussion

drdanz, 2015-11-18:
At the moment we are using this versioning convention: ``MAJOR.MINOR.PATCH.TWEAK``.

* ``MAJOR`` is always 2 and is probably not going to change until we re-write YARP from scratch.
* ``MINOR`` is 3 since 2010. Version 4 was planned long time ago, and now that the plugins are fully working, i believe that the last feature we want, is to get rid of GTK.
* ``PATCH`` Was bumped to 60+ when we decided to break compatibility with the old ``2.3.x`` releases, when we decided that ``2.3.22`` was stable and the newer versions unstable. I'm no longer sure that this was needed, but anyway we cannot turn back now
* ``TWEAK`` Is never tagged, and it is the most confusing. Usually the last number is used for small fixes that do not break compatibility or add new functionality. We are using it in the exact opposite way, i.e. we bump it when a new feature is available, or when we break compatibility with something.

We are tagging versions with a very slow cycle. These are the releases since 2013:

* 2 releases in Jul 2013: the second one is probably a bugfix release
```
v2.3.21 Mon Jul 8 18:17:19 2013 +0200 yarp release 2.3.21 
v2.3.22 Fri Jul 26 14:49:27 2013 +0200 yarp release 2.3.22 
```

* 3 releases within 20 days in Jan/Feb 2014: the second and the third ones are probably bugfix releases
```
v2.3.60 Wed Jan 29 10:48:03 2014 +0100 yarp release 2.3.60 
v2.3.61 Fri Jan 31 15:33:11 2014 +0100 yarp release 2.3.61 
v2.3.62 Wed Feb 19 13:58:39 2014 +0100 yarp release 2.3.62 
```

* 1 release in Jul 2014:
```
v2.3.63 Mon Jul 14 12:40:36 2014 +0200 YARP release 2.3.63 
```

* 1 release in Jun 2015:
```
v2.3.64 Mon Jun 15 12:06:27 2015 +0200 YARP release 2.3.64 
```

I believe the *name* ``TWEAK`` is very misleading. We never tag it, therefore people who require a specific version are forced to use master. For example iCub requires YARP 2.3.64.8. There is no way for git to know whether you are using a compatible version, and this is a problem for YCM Superbuilds.

--------

My suggestions:

1. Make a tag now and whenever we increase the ``TWEAK`` number. Instead of tagging ``v2.3.63.8`` I suggest we tag ``v2.3.63+dev8``, or ``v2.3.64~preview8``, or ``v2.3.64~alpha8``, or something similar, so that it is obvious from the tag that this is not a tweak release, but something that contains new versions. Perhaps until we release YARP 2.4.0 we should change the word ``TWEAK``.

2. Release YARP 2.4.0 as soon as possible. We should make a review of the issues and move to the milestone ``YARP 2.4.0`` all the issues that we want fixed, and then fix them. Hopefully the Qt5 version of the builder will be available soon enough, so that we can get rid of the GTK stuff.

3. Increase the release cycle speed. I believe we should aim at having 3 feature releases per year, therefore a new release every 4 months. People using binaries should not be missing hundreds of bug fixes and new features. We should also have stable releases whenever there is the need to have one. We should have a stable and development branches, and commit bugfixes in the stable branch and new features and not compatible changes (including changes that break the ABI) in the development one. We should aim at keeping full compatibility in newer versions, and mark features as deprecated instead of removing them, perhaps with some specific policy (e.g. remove deprecated functions after 3 feature releases).

4. Starting with YARP 2.4.0, I propose this convention:
 * ``MAJOR`` (2) probably never increased
 * ``MINOR`` (4) increased at every feature release
 * ``PATCH`` (0) increased at every stable release.

 New features can be marked using the ``PATCH`` number, with the extra convention to use a very large number (i.e. +60) for developer versions, i.e. ``v2.4.0``, ``v2.4.1``, etc. will be the stable releases that will be packaged for all the supported platforms, ``v2.4.60`` ``v2.4.61`` will be the developer releases, that will be tagged, but not necessarily packaged (maybe just the source code).

 In this way ``PATCH`` is really for "patch" releases, i.e. it does not break things or contain new features.

@lornat75 what do you think?

traversaro, 2015-11-18:
`v2.3.63+dev8` will still have (at CMake level) `YARP_FIND_VERSION_TWEAK` set to 8? That would be really convenient for enforcing that a given bug is fixed in the `find_package` of the downstream project.

drdanz, 2015-11-18:
@traversaro Yes, my plan is to have a tag (e.g. ``v2.3.63+dev8``) that corresponds to what can be found using, but is not ``v2.3.63.8`` because that's in my opinion really confusing. What do you think?

```cmake
find_package(YARP 2.3.63.8)
```

traversaro, 2015-11-20:
I am dealing with version numbers most at the CMake level, so it is quite natural for me that that the CMake version `2.3.63.8` correspond to tag `v2.3.63.8` . I guess that for people more used to other versioning systems (Debian packages?) that could be confusing, but that not the case for people that deal with version mostly at the CMake level.

traversaro, 2015-11-20:
Some poor man comments to @drdanz suggestions: 

> Perhaps until we release YARP 2.4.0 we should change the word TWEAK.

TWEAK is part of the CMake jargon https://cmake.org/cmake/help/v3.0/command/find_package.html , why we should change it? 

> We should have a stable and development branches, and commit bugfixes in the stable branch and new features and not compatible changes (including changes that break the ABI) in the development one.

This reminds me of https://github.com/robotology/gazebo-yarp-plugins/issues/112 . 

My impression is that if we'll go for the separate "stable" and "development" branches what will happen in practice is that everyone at IIT (and research groups more active/close to IIT) will simply track the development branch (as they now track the master branch). For "lazy" users, if feature release are sufficiently frequent maintaining old feature release is not really worth, "lazy" users can simply wait for the next feature release.

I think increasing the frequencies of feature releases makes a lot of sense, but that is only possible if making a release is not a task involving a lot of work (I remember that @lornat75 mentioned that doing a release is now approximately 2 man days). Backporting bug fixes to "release" branches, could make sense, but I think it is worth the trouble only if some users of an old release explicitly requests the bug fix (this is similar to the strategy that they use in Gazebo).

lornat75, 2015-11-25:
I agree we should increase the release cycle. I agree we should tag a 2.3.4 as soon as possible.

We do need a way to ensure that CMake can detect changes in YARP at a much finer granularity than releases. This feature is helpful for detecting changes in YARP that are required by dependencies (i.e. icub-main), This was the intended use of TWEAK. TWEAK does not go to releases because TWEAK should be set to 0 at every new release (which increases the MINOR or PATCH).

I agree with @traversaro that moving to "a development"  branch will mean that all active users will start tracking it with minor improvement in the development cycle. It will have the benefit that many users will get slower updates from master but it requires duplicating the CI infrastructure.

We can discuss these issues.

drdanz, 2015-11-25:
> My impression is that if we'll go for the separate "stable" and "development" branches what will happen in practice is that everyone at IIT (and research groups more active/close to IIT) will simply track the development branch (as they now track the master branch).

I agree, my suggestion is to have master as development branch, and make stable branches when we make releases, i.e. `yarp-2.4` could be the stable branch for the 2.4 release, and we could tag v2.4.1 on that branch.

> I guess that for people more used to other versioning systems (Debian packages?) that could be confusing, but that not the case for people that deal with version mostly at the CMake level. 

The word "TWEAK" to track new features is in my opinion very misleading for anyone who has a conception of version.
Another option could be to use the date to track new features (for example `v2.3.63.20151125`) CMake internally does something similar, see [CMakeVersion.cmake](https://github.com/Kitware/CMake/blob/master/Source/CMakeVersion.cmake), [CMakeVersionCompute.cmake](https://github.com/Kitware/CMake/blob/master/Source/CMakeVersionCompute.cmake), and [CMakeVersionSource.cmake](https://github.com/Kitware/CMake/blob/master/Source/CMakeVersionSource.cmake). They don't have tags for these, though, also because the date is increased every day by a bot...
In YCM I'm using something similar, but with the "debian" concept of "~" in the version number i.e. ``0.2.0~20151125 < 0.2.0``. By the way is not compatible with git (you cannot use ~ in tags) or CMake (cannot use ~ in find_package), see also robotology/ycm#81
Another option is to use odd-numbered versions for development releases like many projects do. I don't like it very much, but it works and it is compatible both with CMake and git

traversaro, 2015-12-18:
> I agree, my suggestion is to have master as development branch, and make stable branches when we make releases, i.e. yarp-2.4 could be the stable branch for the 2.4 release, and we could tag v2.4.1 on that branch.

In this case, what will be the policy for bug fixes?

drdanz, 2016-06-20:
The new even-odd numbering policy fixes this issue. Closing

closed 2016-06-20 by drdanz