robotology/yarp #641: running yarpdataplayer in Mac os x

zahrag, 2015-11-18:
In Mac Os X, I have Yarp installed from source and tested and it works. When I run yarpdataplayer, I have this error:

dyld: Library not loaded: /usr/local/lib/libavcodec.56.dylib
  Referenced from: /Users/zahra/Desktop/Robotology/opencv-3.0.0/build/lib/libopencv_highgui.3.0.dylib
  Reason: image not found
Trace/BPT trap: 5

Does anybody know how to fix it?

best,
zahra

kt10aan, 2015-11-19:
Since you are using OpenCV outside the normal library paths, you can use ``INSTALL_WITH_RPATH ON`` in CMake.

Alternatively, you can install OpenCV using ``Homebrew``

gsaponaro, 2015-11-19:
Recently (https://github.com/robotology/yarp/commit/ac69d1b7d27d99c274d849c21d40ee4aa4f36595), RPATH was turned on by default in yarp.

zahrag, 2015-11-19:
yep, and it did not help!

kt10aan, 2015-11-19:
Actually the problem is not with YARP. It is with how you compiled OpenCV. It is OpenCV that cannot find libavcodec.

zahrag, 2015-11-19:
I installed Open CV, 

brew tap homebrew/science
brew install opencv3

BUT, when I want to install yarp from source, when use "make" command I receive:
-------------------------- Error--------------------------------
In file included from /Users/zahra/Desktop/Robotology/yarp/extern/goocanvas/goocanvas/src/goocanvas.c:98:
/usr/local/include/glib-2.0/glib/gi18n-lib.h:23:10: fatal error: 'libintl.h' file not found
#include <libintl.h>
         ^
1 error generated.
make[2]: *** [extern/goocanvas/CMakeFiles/YARP_priv_goocanvas.dir/goocanvas/src/goocanvas.c.o] Error 1
make[1]: *** [extern/goocanvas/CMakeFiles/YARP_priv_goocanvas.dir/all] Error 2
make: *** [all] Error 2
------------------------------------------------------------------------------------------------------------

?!!

kt10aan, 2015-11-19:
Technically, thats another issue, not the same one :). Anyway, if you have installed ``gettext`` with ``Homebrew``, the file should be in ``/usr/local/Cellar/gettext/0.19.x/include/libintl.h`` and make a link to it in ``/usr/local/include``.

zahrag, 2015-11-19:
I reinstalled the yarp successfully from the source, and Open Cv too, but the same error exists, :D, 
anyway, thank u, :)

francesco-romano, 2016-09-01:
@zahrag any update on this?

pattacini, 2016-09-01:
I think we can close this issue.

closed 2016-09-01 by pattacini