robotology/yarp #642: yarpmotorgui misbehavior

jeljaik, 2015-11-19:
I have noticed (and confirmed also by @francesco-romano, that on OS X the yarpmotorgui behavior differs from the one on Linux (at least in the Position Control Mode). There are two differences:

1. Moving the slider sends immediately the references to the position controllers as it can be seen in the following gif :arrow_down: Notice how the encoders follow the slider almost immediately. 

[image]

I personally find this behavior a bit dangerous and annoying.

2. After focusing the sliding, the arrow keys do not have any effect.

pattacini, 2015-11-19:
+100 for the animated gif :smirk: 
@randaz81 is working on a new release of `yarpmotorgui` with brand new widget that should standardize the use of the slider. It should be merged quite soon.
Stay tuned.

jeljaik, 2015-11-19:
:bowtie: Great, thanks!

barbalberto, 2015-11-20:
Actually I have the same behaviour on Linux also.

jeljaik, 2015-11-20:
On my computer and @francesco-romano's we last updated Yarp on November 11, same for the machine with Ubuntu 14.04. So ... at least until that date, the behaviors were different.

randaz81, 2017-01-10:
@francesco-romano @jeljaik Can you check if this is a still an issue?