robotology/yarp #644: Make yarpmanager perform a dependency check automatically when opening an application XML

gsaponaro, 2015-11-23:
Currently, when I launch **yarpmanager** and I open an application, all dependencies appear as red/unmet, regardless of them being "really" down or up. I have to do a manual refresh to update the dependencies list and see what's already running.

Suggestion:

It would be nice if, upon application opening, the program was smart enough to notice which dependencies were on already. It would save 1 click... multiplied by the (high) number of times that this task is performed by developers. :)

Example:

[image]

apaikan, 2017-02-22:
Hi @gsaponaro 

Your suggestion seem quite rational to me :) 

But there is a `tiny` reason  why 'automatic refreshing' has been avoided:

- `refresh` (opposed to `connect`, `run`, ..) is the only action in the `yarpmanager` which is `NOT` threaded (i.e. is a blocking action). This is not an implementation constraint but mostly the behavioral constraint. 
- So what it has to do with automation? there might be some of the resources, ports,  (e.g. `/icub-laptop`)  are in `zombie` states due to some physical network link problem, crashing and etc. These abnormal  ports may block the `refresh` and as result blocks the application TAB in GUI while loading the application. Thus user cannot see/perceive  what is happening and where (e.g. which port) is the cause of the problem
-  **One more click** costs less than deal with all these :see_no_evil: 

Now, If my explanation also seems rational to you, go ahead to close the issue :)

gsaponaro, 2017-02-22:
> Now, If my explanation also seems rational to you, go ahead to close the issue :)

Yes it does, thanks for taking the time to explain this. I guess we'll have to continue doing that click, then. No problem. :)

closed 2017-02-22 by gsaponaro