robotology/yarp #646: Proposal: include Eigen in YARP Windows installer

traversaro, 2015-11-24:
We are currently working to simplify use (and thus dissemination) of  the CoDyCo/Koroibot software. As part of that effort, we are removing a lot dependencies on our software: for example we are working to make the iDynTree library (the library with all the whole body kinematics/dynamics code) work with just two dependencies (just for building, headers don't include any "foreign" header): TinyXML and [Eigen](http://eigen.tuxfamily.org). Getting this dependencies installed is not a problem on Linux or OS X, but it always problematic for Windows users, especially for not expert ones.
While TinyXML is small enough (and not updated) so that it can be included directly in the repository, Eigen is a frequently updated library of which it would be cumbersome to keep a copy in the repo. 

I was then wondering if it could make sense to add Eigen to the YARP dependencies installer: Eigen is an header only library of ~3MB , while the installer is already huge (~400 MB) so the increased size should not be a big problem. While being a convenience for us (and also for a lot of other people using Eigen) it could also be a first step toward resolving : https://github.com/robotology/yarp/issues/266 , as Eigen could be used to substituted GSL in the libYARP_math implementation. It you are ok with this, I would be happy to modify the script for producing the NSIS installer.

@lornat75 @mbrunettini @drdanz

lornat75, 2015-11-25:
Look reasonable to me.

drdanz, 2015-11-25:
+1
Should be a separate issue, but I'll just leave it here... we should also add `libjpeg` for the mjpeg carrier

Good luck with the NSIS installer :sob:

traversaro, 2016-12-20:
Done since the binaries of `v2.3.68`, see https://github.com/robotology/yarp/releases/tag/v2.3.68 .

closed 2016-12-20 by traversaro