robotology/yarp #649: run yarpview from matlab

wainting, 2015-11-26:
Hi to all,
I hope to be in the right place to make this question. I use this guide (http://wiki.icub.org/wiki/Calling_yarp_from_Matlab) to call yarp from matlab. All the things work but my goal is to run yarpview from matlab.  I can't find a yarpview.java to call from matlab and my question is if exists one or if there is an alternative way to make matlab "see" what appears on yarpview cameras. Thanks

Best 

Daniele

pattacini, 2015-11-26:
`yarpview` is a binary so I thinks it's not the correct way of displaying images within MATLAB.

You should retrieve images from yarp ports and then somehow convert them in a MATLAB compatible format, which is suitable for display with plot related commands.

wainting, 2015-11-26:
I've found the function FastImageConversion.m  inside the example directory of yarp  and I'm trying to use it for my task . Maybe it's better open a new issue here https://github.com/robotology/yarp/blob/master/example/matlab/FastImageConversion.m or move this one.

pattacini, 2015-11-26:
@wainting, issues are not attached to files :smirk: 
we can keep tracking it here.

wainting, 2015-11-26:
Ok I solve my problem :+1: .
I've simply follow these lines from FastImageConversion.m:

% run the example
% connect images to input - e.g. yarp connect /icub/camcalib/left/out /matlab/img:i
% connect images to output to yarpview - e.g. yarp connect /matlab/img:o /viewOut
% connect write port to start and quit - e.g. yarp write ... /matlab/read
 %
 % use any type of message to start eg. "go"
 % when convertion is finished type "quit" to close ports nicely

Instead of /icub/camcalib/left/out i use:

 yarpview --name /view/right 
yarp connect /icubSim/cam/left /view/left    (after running iCub_SIM)
yarp connect /icubSim/cam/right /matlab/img:i 

and finally i obtain my frames of the right camera in a matlab matrix.

pattacini, 2015-11-26:
Great. Shall we close this issue then?

wainting, 2015-11-26:
Sure. Thanks Ugo.

closed 2015-11-26 by wainting