robotology/yarp #653: Error when running from the yarp manager

robertocalandra, 2015-12-03:
Hello,

I am trying to run some old application I had on yarp manager. However when trying to run it, some of the programs do not start and instead return an error. For example yarpmotorgui or yarpview.

I am sure that the application used to work. However, what is even more strange is that if I run exactly the same commands from the terminal (with all the arguments), the programs start normally.

Any suggestion on what could be the cause?

lornat75, 2015-12-04:
Hi Roberto,
can you attach the xml(s) you are trying to run?

@apaikan: can you check this?

robertocalandra, 2015-12-04:
Here it is:

```
<application>
<name>iCubTelecontrol_highres</name>

        <dependencies>
        </dependencies>
  
        
        <module>
                <name>yarpmotorgui</name>
                <!-- Remember to add the "--headV2" parameter for robots equipped with head v. 2.0 -->
                <parameters>--headV2</parameters>
                <node>tomlinson</node>
                <tag>yarpMotorGuiTag</tag>
        </module>

<!-- CAMERA -->

<module>
  <name>icubmoddev</name>
      <parameters> --from camera/dragonfly2_config_left_bayer_640_480.ini</parameters>
  <node>pc104</node>
  <tag>camera_left</tag>
</module>
<module>
  <name>icubmoddev</name>
      <parameters> --from camera/dragonfly2_config_right_bayer_640_480.ini</parameters>
  <node>pc104</node>
  <tag>camera_right</tag>
</module>
<module>
  <name>camCalib</name>
  <parameters>--context cameraCalibration --from icubEyes_640x480.ini --group CAMERA_CALIBRATION_LEFT --name /icub/camcalib/left</parameters>  
  <node>tomlinson</node>
  <stdio>tomlinson</stdio>
  <tag>camCalibLeft</tag>
</module>
<module>
  <name>camCalib</name>
  <parameters>--context cameraCalibration --from icubEyes_640x480.ini --group CAMERA_CALIBRATION_RIGHT --name /icub/camcalib/right</parameters> 
  <node>tomlinson</node>
  <stdio>tomlinson</stdio>
  <tag>camCalibRight</tag>
</module>

<module>
  <name>yarpview</name>
  <parameters>--name /icub/view/left --x 0 --y 0 --p 50</parameters>
  <node>tomlinson</node>
  <tag>viewleft</tag>
</module>

<module>
  <name>yarpview</name>
  <parameters>--name /icub/view/right --x 320 --y 0 --p 50</parameters>
  <node>tomlinson</node>
<tag>viewright</tag>
</module>

<module>
  <name>frameGrabberGui2</name>
  <parameters>--local /icub/fggui/left --remote /icub/cam/left --x 0 --y 350 --width 320 --height 500</parameters>
  <tag>fggui_left</tag>
  <node>tomlinson</node>
</module>

<module>
  <name>frameGrabberGui2</name>
  <parameters>--local /icub/fggui/right --remote /icub/cam/right --x 320 --y 350 --width 320 --height 500</parameters>
  <tag>fggui_right</tag>
  <node>tomlinson</node>
</module>

<connection>
  <from>/icub/cam/left</from>
  <to>/icub/camcalib/left/in</to>
  <protocol>udp+recv.bayer+method.nearest</protocol>
</connection>
<connection>
  <from>/icub/cam/right</from>
  <to>/icub/camcalib/right/in</to>
  <protocol>udp+recv.bayer+method.nearest</protocol>
</connection>
<connection>
  <from>/icub/camcalib/left/out</from>
  <to>/icub/view/left</to>
  <protocol>udp</protocol>
</connection>
<connection>
  <from>/icub/camcalib/right/out</from>
  <to>/icub/view/right</to>
  <protocol>udp</protocol>
</connection>

<!-- SKIN -->

	<module>
		<name>iCubSkinGui</name>
		<!-- Remember to use the proper configuration files (i.e. "_V2.ini" for V2 robots) -->
		<parameters>--from right_hand_V2.ini --useCalibration --xpos 960 --ypos 0 --width 300 --height 300</parameters>
		<node>tomlinson</node>
		<tag>skinGuiRH</tag>
	</module>

	<module>
		<name>iCubSkinGui</name>
		<!-- Remember to use the proper configuration files (i.e. "_V2.ini" for V2 robots) -->
		<parameters>--from left_hand_V2.ini --useCalibration --xpos 320 --ypos 0 --width 300 --height 300</parameters>
		<node>tomlinson</node>
		<tag>skinGuiLH</tag>
	</module>

	<module>
		<name>skinManager</name>
		<parameters>--from skinManAll.ini --context skinGui</parameters>
		<node>tomlinson</node>
		<tag>skinMan</tag>
	</module>
	<module>
		<name>skinManagerGui</name>
		<parameters>--from skinManGui.ini --context skinGui --xpos 1020 --ypos 350</parameters>
		<node>tomlinson</node>
		<tag>skinManGui</tag>
	</module>

  <connection>
    <from>/skinManagerGui/rpc:o</from>
    <to>/skinManager/rpc</to>
    <protocol>tcp</protocol>
  </connection>

  <connection>
    <from>/skinManager/monitor:o</from>
    <to>/skinManagerGui/monitor:i</to>
    <protocol>udp</protocol>
  </connection>

  <connection>
    <from>/skinManager/info:o</from>
    <to>/skinManagerGui/info:i</to>
    <protocol>tcp</protocol>
  </connection>

  <connection>
    <from>/icub/skin/left_hand_comp</from>
    <to>/skinGui/left_hand:i</to>
    <protocol>udp</protocol>
  </connection>

  <connection>
    <from>/icub/skin/right_hand_comp</from>
    <to>/skinGui/right_hand:i</to>
    <protocol>udp</protocol>
  </connection>

</application>
```

pattacini, 2015-12-04:
@robertocalandra 
As side note, concerning **`icubmoddev`**, please take https://github.com/robotology/QA/issues/68 into account and replace any occurrence of it with **`yarpdev`**.

lornat75, 2015-12-08:
It looks like you are having problems running guis on "tomlinson"; is this correct?

How do you run yarprun on this machine? Do you use icubcluster.py? In this case did you add the "display" property, in the cluster-config.xml file, i.e as follows:

    <node display=":0.0">tomlinson</node>

lornat75, 2015-12-10:
An intermediate debugging step is the following:

Try running yarprun at the terminal on tomlinson:

  yarprun --server /tomlinson

Make sure yarprun is not already running before doing this (for example from icubcluster.py).

Try running the guis again from the yarpmanager. If they run, the problem is in the configuration of icubcluster orthe ssh daemon on tomlinson; we can focus on fixing these.

lornat75, 2016-01-12:
If you are running yarprun with icub-cluster see this issue:

https://github.com/robotology/icub-main/issues/272

lornat75, 2016-01-12:
I am closing this issue. If problems persists please re-open it.

closed 2016-01-12 by lornat75