robotology/yarp #657: Kernel Panic on yarp connect

nico202, 2015-12-10:
Hi, I'm running yarp 2.3.63.8, kernel version 4.4.0-040400rc4-generic (happened with rc2 too, I'll try with other version as soon as I can to find the regression).
If I run `yarpserver` and then in another terminal emulator `yarp connect` I get a Kernel Panic. I don't know how to provide a backtrace for this, I'll follow istruction if someone can provide a link. (also, don't know if I have to report this here or on the kernel repo)
Thanks, Nicolò

lornat75, 2015-12-11:
Dear nico22, it seems quite unlikely that yarp can cause kernel panic, I would blame a malfunctioning driver (the network card?). Are you sure your system is sane?

nico202, 2015-12-11:
Quite sure, works well on 4.3 and older. Also, it's the only situation 
that triggers it.
If can run some test to be sure the system is sane I'll run it.
Thanks, Nicolò

lornat75, 2016-02-04:
Closing as no more input was received. Feel free to re-open again.

closed 2016-02-04 by lornat75