robotology/yarp #658: YARP bindings do not compile (again?!?)

arocchi, 2015-12-10:
Setting `YARP_COMPILE_BINDINGS:BOOL=ON` and `CREATE_PYTHON:BOOL=ON` results in the following error:

```
[100%] Building CXX object bindings/CMakeFiles/_yarp_python.dir/yarpPYTHON_wrap.cxx.o
/home/walkman/robotology-superbuild/build/external/YARP/bindings/yarpPYTHON_wrap.cxx: In function ‘PyObject* _wrap_PortReaderBufferBase_setAutoRelease__SWIG_0(PyObject*, PyObject*)’:
/home/walkman/robotology-superbuild/build/external/YARP/bindings/yarpPYTHON_wrap.cxx:33500:32: warning: ‘void yarp::os::impl::PortReaderBufferBase::setAutoRelease(bool)’ is deprecated (declared at /home/walkman/robotology-superbuild/external/YARP/src/libYARP_OS/include/yarp/os/PortReaderBuffer.h:275) [-Wdeprecated-declarations]
     (arg1)->setAutoRelease(arg2);
                                ^
/home/walkman/robotology-superbuild/build/external/YARP/bindings/yarpPYTHON_wrap.cxx: In function ‘PyObject* _wrap_PortReaderBufferBase_setAutoRelease__SWIG_1(PyObject*, PyObject*)’:
/home/walkman/robotology-superbuild/build/external/YARP/bindings/yarpPYTHON_wrap.cxx:33525:28: warning: ‘void yarp::os::impl::PortReaderBufferBase::setAutoRelease(bool)’ is deprecated (declared at /home/walkman/robotology-superbuild/external/YARP/src/libYARP_OS/include/yarp/os/PortReaderBuffer.h:275) [-Wdeprecated-declarations]
     (arg1)->setAutoRelease();
                            ^
/home/walkman/robotology-superbuild/build/external/YARP/bindings/yarpPYTHON_wrap.cxx: In function ‘PyObject* _wrap_ResourceFinder_getContextPath(PyObject*, PyObject*)’:
/home/walkman/robotology-superbuild/build/external/YARP/bindings/yarpPYTHON_wrap.cxx:40193:37: warning: ‘yarp::os::ConstString yarp::os::ResourceFinder::getContextPath()’ is deprecated (declared at /home/walkman/robotology-superbuild/external/YARP/src/libYARP_OS/include/yarp/os/ResourceFinder.h:246) [-Wdeprecated-declarations]
     result = (arg1)->getContextPath();
                                     ^
/home/walkman/robotology-superbuild/build/external/YARP/bindings/yarpPYTHON_wrap.cxx: In function ‘PyObject* _wrap_IGazeControl_getSaccadesStatus(PyObject*, PyObject*)’:
/home/walkman/robotology-superbuild/build/external/YARP/bindings/yarpPYTHON_wrap.cxx:74887:28: error: ‘class yarp::dev::IGazeControl’ has no member named ‘getSaccadesStatus’
     result = (bool)(arg1)->getSaccadesStatus(arg2);
                            ^
/home/walkman/robotology-superbuild/build/external/YARP/bindings/yarpPYTHON_wrap.cxx: In function ‘PyObject* _wrap_IGazeControl_setSaccadesStatus(PyObject*, PyObject*)’:
/home/walkman/robotology-superbuild/build/external/YARP/bindings/yarpPYTHON_wrap.cxx:76131:28: error: ‘class yarp::dev::IGazeControl’ has no member named ‘setSaccadesStatus’
     result = (bool)(arg1)->setSaccadesStatus(arg2);
                            ^
make[6]: *** [bindings/CMakeFiles/_yarp_python.dir/yarpPYTHON_wrap.cxx.o] Error 1
make[5]: *** [bindings/CMakeFiles/_yarp_python.dir/all] Error 2
make[4]: *** [all] Error 2
make[3]: *** [external/YARP/CMakeFiles/YCMStamp/YARP-build] Error 2
make[2]: *** [CMakeFiles/YARP.dir/all] Error 2
make[1]: *** [CMakeFiles/YARP.dir/rule] Error 2
make: *** [YARP] Error 2
```

Any clues @barbalberto @apaikan @drdanz ? 
@liesrock what do you think?

pattacini, 2015-12-10:
Is `yarp` up-to-date or by any chance is there any leftover in the cache?

arocchi, 2015-12-10:
@pattacini honestly I am getting two different behaviors from different computers. On my personal laptop everything working (and has kept working since some time), on one of the simulator servers and some other laptops in the lab I am getting this error, with the latest master (9b2c2d7e) and with a full delete of the `build/` folder, with:

```
CMAKE_CACHE_ARGS -DCREATE_IDLS:BOOL=ON
                                 -DCREATE_SHARED_LIBRARY:BOOL=ON
                                 -DYARP_USE_SYSTEM_SQLITE:BOOL=ON
                                 -DCREATE_LIB_MATH:BOOL=ON
                                 -DYARP_USE_ATLAS:BOOL=OFF
                                 -DCREATE_OPTIONAL_CARRIERS:BOOL=ON
                                 -DENABLE_yarpcar_bayer_carrier:BOOL=ON
                                 -DUSE_LIBDC1394:BOOL=OFF
                                 -DENABLE_yarpcar_tcpros_carrier:BOOL=ON
                                 -DENABLE_yarpcar_xmlrpc_carrier:BOOL=ON
                                 -DENABLE_yarpcar_priority_carrier:BOOL=ON
                                 -DENABLE_yarpcar_bayer_carrier:BOOL=ON
                                 -DENABLE_yarpidl_thrift:BOOL=ON
                                 -DCREATE_DEVICE_LIBRARY_MODULES:BOOL=ON
                                 -DENABLE_yarpcar_human_carrier:BOOL=ON
                                 -DENABLE_yarpcar_mjpeg_carrier:BOOL=ON
                                 -DENABLE_yarpcar_rossrv_carrier:BOOL=ON
                                 -DENABLE_yarpmod_fakebot:BOOL=ON
                                 -DENABLE_yarpmod_opencv_grabber:BOOL=OFF
                                 -DYARP_CLEAN_API:BOOL=ON
                                 -DYARP_COMPILE_TESTS:BOOL=OFF
                                 -DYARP_COMPILE_EXPERIMENTAL_WRAPPERS:BOOL=ON
                                 -DYARP_COMPILE_BINDINGS:BOOL=ON
                                 -DCREATE_PYTHON:BOOL=ON
                                 -DCREATE_GUIS:BOOL=ON
```

any ideas on where should I start troubleshooting?

arocchi, 2015-12-10:
solved with the interventino of @apaikan. @pattacini all the cases were it did not work were of computers with `yarp` installed at the system level. The local (YCM/superbuild) installation of YARP would generate the bindings using the system level yarp includes, and that would result in errors.
We fixed by removing the superfluous system-level installation.
Maybe an idea could be to force the use of the local header rather than the installed ones?

traversaro, 2015-12-10:
Definitely @arocchi , it is something that we agreed for a long time, see https://github.com/robotology/yarp/tree/disable-bindings-with-external-yarp . 
Can you check if this commit fix your issue: https://github.com/robotology/yarp/compare/disable-bindings-with-external-yarp ? Thanks.

arocchi, 2015-12-10:
@traversaro unfortunately that commit (for some reason..) does not solve the issue... yet in the modified `CMakeLists.txt`,  `YARP_INCLUDE_DIRS` correctly refers to the local `include` folders....

traversaro, 2015-12-10:
Probably some other `include_directories` command is including the system path (`/usr/include` or `/usr/local/include`) in which the system yarp includes are installed, but this is strange because apparently the first `include_directory` command. 
To debug this you can run `make VERBOSE=1` to see the actual order of the include directories passed to the swig tool. If for some reason the yarp local include folders are not the first, you should be able to force them to be the first one by adding a `BEFORE` to the `include_directories command:
~~~
include_directories(BEFORE ${YARP_INCLUDE_DIRS})
~~~

arocchi, 2015-12-11:
@traversaro the local `${YARP_INCLUDE_DIRS}` are included in the right order  when compiling the bindings library with `c++  ... yarpPYTHON_wrap.cxx`. The problem is that when generating that file from `yarp.i`  wrappers for functions that are present in the system level yarp installation are created, but then during the compilation phase they are not found. We should find a way the local headers when calling `swig`

traversaro, 2015-12-11:
The swig tool should use the same include directories passed to the compiler. However you can verify this which directories are actually passed to swig and which one to the compiler by running `make VERBOSE=1`.

arocchi, 2015-12-11:
Yes, while the `c++` command invocation doesn't have `/usr/include`, the `swig` invocation has, because of `include_directories(${PYTHON_INCLUDE_PATH})`
Specifying `include_directories(BEFORE ${YARP_INCLUDE_DIRS})` solves the issue

traversaro, 2015-12-13:
:+1: 

I think we then just need to cleanup the documentation that is still referring to the legacy way of compiling bindings, and then we could merge those changes.

drdanz, 2015-12-22:
Whenever we are removing the possibility to build bindings out of the build tree, we should also remove the `` YARP_COLLISION_AVOIDANCE`` madness, and just make it always enabled, eventually fixing the ``OUTPUT_NAME`` properties and/or output directories.

drdanz, 2016-01-04:
@traversaro I rebased and added a commit to your `disable-bindings-with-external-yarp` branch, removing the `YARP_COLLISION_AVOIDANCE` and making the cmake code for the bindings *almost* readable, I think that as soon as we clean the documentation about the out-of-source build for the bindings, we can merge the branch. We should still ensure that nobody is using it, though.

closed 2018-12-07 by arocchi