robotology/yarp #659: Deprecate void yarp::os::Bottle::add(const char* txt)

drdanz, 2015-12-16:
According to the [comments in the code](https://github.com/robotology/yarp/blob/master/src/libYARP_OS/include/yarp/os/Bottle.h#L140)

```c++
    // worked "accidentally", user code relies on this
    void add(const char* txt) {
        addString(txt);
    }
```

What do you think about deprecating this function in favour of ``yarp::os::Bottle::addString(const char* txt)``?

At least inside yarp I'm planning to replace all the calls with ``addString()``, since using this method just adds an extra function call...

@lornat75, @pattacini Do you have any idea about how much impact deprecating this method will have on the user code? (Note that it's not removed, just deprecated therefore it will just cause an extra warning if warnings are enabled).

pattacini, 2015-12-16:
Fine with me :+1:

lornat75, 2015-12-16:
Fine with me.

commit 31bdd9ab (2016-01-04): Cleanup yarp::os::Bottle

Properly implement PIMPL idiom, hiding all private members.

Deprecate void yarp::os::Bottle::add(const char* txt) (Closes #659)

WARNING; <yarp/os/Value.h> is no longer included in Bottle.h, therefore
         in some places this could break the build.
closed 2016-01-04 by drdanz