robotology/yarp #664: Java bindings fail to build

Tobias-Fischer, 2016-01-06:
Hi,
since the latest updates to yarp, the java bindings do not build anymore. The errors are:
- yarp/build/bindings/yarpJAVA_wrap.cxx:11814: error: 'class yarp::os::Bottle' has no member named 'specialize'
- yarp/build/bindings/yarpJAVA_wrap.cxx:11842: error: 'class yarp::os::Bottle' has no member named 'setNested'
- yarp/build/bindings/yarpJAVA_wrap.cxx:20279: error: 'ConnectionReader' is not a member of 'yarp'
- yarp/build/bindings/yarpJAVA_wrap.cxx:20301: error: 'ConnectionWriter' is not a member of 'yarp'

Some of them occur several times.

Best, Tobi

drdanz, 2016-01-06:
Have you tried removing the `yarpJAVA_wrap.cxx` file?

traversaro, 2016-01-06:
Yes, the problem is similar to https://github.com/robotology/idyntree/issues/84 . 
The headers are modified, but as the headers are not a dependency of the target for the generation of the `yarpJAVA_wrap.cxx` file, the `yarpJAVA_wrap.cxx` file is not regenerated. 
Another workaround is (besides deleting the `yarpJAVA_wrap.cxx` file) to force regeneration of the swig bindings by pretend that a file on which the `yarpJAVA_wrap.cxx` depends is modified (a simple `touch bindings/yarp.i` is sufficient for this). 

Relevant CMake bug : http://public.kitware.com/Bug/view.php?id=4147 .

Tobias-Fischer, 2016-01-07:
Working fine, thanks for the hint @drdanz and @traversaro :)

closed 2016-01-07 by Tobias-Fischer