robotology/yarp #674: add velocity controller with additional position reference

kjyv, 2016-02-04:
Hey,

In order to create smooth trajectories for inertia parameter identification, I'd like to use a velocity controller which also has position input, basically a p-pi cascaded controller. The existing position or velocity controllers each have drawbacks for tracking a trajectory with a velocity profile. I figure from the source that since controllers are implemented in YARP directly, I'd have to add such an implementation next to the existing controllers. Would that be the preferred way?

pattacini, 2016-02-04:
Hi @kjyv 

As quick answer, I'd suggest you to integrate your high level velocity profiles to get position references that in turn you can send to the robot relying on the `IPositionDirect` control.

The resulting trajectories performed by the robot will be pretty smooth (we use this technique quite often).

randaz81, 2016-02-04:
> I figure from the source that since controllers are implemented in YARP directly [...]

This is not true. Yarp only provides a convenient set on interfaces and methods, but the implementation is system-dependent and actually resides in the robot firmware (or in the simulator source code)...

kjyv, 2016-02-04:
Hey, thanks for the suggestion. I've since seen that Yarp only provides the interfaces. Even those are a bit specific though for a fixed set of different types of controllers and does not include the one I need.
So I would change my question then to say adding an interface for the controller I need.

@pattacini: I've been sending positions already and they come from smooth harmonic functions (so I have positions and velocity). The problem it seems is that the position controller has varying speeds depending on the error to the current position and therefore introduces oscillations because it stops at certain points when they are reached before the next points are far enough again. I could potentially also change the existing interfaces to allow setting the max speed but the controller will still not be very good.

iron76, 2016-02-04:
The discussion intrigues me and I would like to know @traversaro opinion. But first, @kjyv, can you be more precise in telling us what kind of trajectories you want to execute that are not feasible with the current YARP-interfaces?

kjyv, 2016-02-04:
@iron76: Sure. So, I want to generate trajectories for parameter identification, which will be coming from Fourier series. For now they are very simple smooth oscillations but depending on optimization criteria can look also quite different of course. However, the resulting positions I measure are full of spikes and also looking at the movement, I see interruptions and oscillations.This is in gazebo so far. Of course there can also be other reasons but the first thought that came to my mind is that instead of a position controller a velocity controller is the better choice. Since it will introduce offsets over time though, an additional p-part in the controller can correct for that. I'm not sure how a pure position controller could properly be used to achieve trajectories with derived from a velocity profile. I guess it's possible to get it generally smooth but it will have wrong velocities at times.

pattacini, 2016-02-04:
@kjyv are you sure you're using the `IPositionDirect` interface? The behavior you described does much resemble the output of a `IPositionControl` interface.

The position references are then tracked by means of PID controllers at low-level (currently), thus the outcome really depends on how those PID are tuned. The use of a PID at low-level, however, does not explain to me the weird behavior you pointed out: that is the reason of my doubt above.

By the way, I do believe that the `IPositionDirect` interface provides the functionality you have in mind. Therefore, eventually, you wouldn't introduce a new interface, but just change its implementation (e.g. with a new controller in place of the current PID).

iron76, 2016-02-04:
Also, I would double check that Gazebo is properly configured for position control. As far as I know it has been mainly used (and hopefully tuned) for torque control and position control might be not perfectly functional. I agree with @pattacini, `IPositionDirect` should be capable of generating chirp-like trajectories.

kjyv, 2016-02-04:
Ok, I wasn't using direct mode, but I'm not sure how this is different from normal position mode except that it does not generate trajectories. In both cases the velocity is dependent on the controller, not of an input that I'd like to give. It will get close to my desired result I guess if I send setpoints quick enough with direct mode, is that what you mean? I've also seen there is VOCAB_CM_MIXED (that I didn't see locally in my older superbuild yarp version...) which sounds pretty much like what I want.
Thanks for the suggestions!

closed 2016-02-04 by kjyv

traversaro, 2016-02-04:
The definite guide (definitly something worth reading!) on YARP control modes is http://wiki.icub.org/wiki/File:ICub_Control_Modes_1_1.pdf . It is linked in http://wiki.icub.org/wiki/Control_Modes , but it is hidden to most users. We should find a better place where to link it.

pattacini, 2016-02-04:
@kjyv `standard position` mode is very very different from `direct mode` :smirk:

The most prominent difference is that in the former mode the user can only specify the final target with the time to reach for it, whereas with the latter the user can specify whatever smooth trajectory is needed to attain for the final point.

Find out more on this reading the guide @traversaro (:+1:) just provided.

pattacini, 2016-02-04:
##### Side note

If you provide a position that varies instant by instant - and the controller underneath is well tuned to track for it - then you're also providing the velocity, which is an information intrinsically embedded in the position reference.

kjyv, 2016-02-04:
Well, I was assuming that standard position mode behaves like direct position control before. I've tried out direct mode and have pretty similar problems. So of course, the velocity is encoded in the position reference but one problem I guess is that the position reference can not be sent at every instant. I would assume the controller still reaches the set point before the next is already set, so the velocity is reduced for a short amount of time. In the retrieved velocity I can see ripples in the otherwise harmonic curve. I'll try mixed mode but I will first have to extend idynutils for that to be available to my module. Thanks!

pattacini, 2016-02-04:
The beauty of the **`direct mode`** is right that you can send fresh set-points each step (a rate of `100 Hz` should be sufficient). It has been deviced for this specific goal indeed.

iron76, 2016-02-04:
@kjyv I understand your concerns and I think you are going into details that might matter in simulation but will never be relevant on real experiments. It seems you are looking for a control mode that tracks a given joint trajectory with zero tracking error even if you don't have a perfect model of the system. Is this the case?

iron76, 2016-02-08:
@kjyv what happened to this issue. Definitively closed?

kjyv, 2016-02-08:
@iron76: sorry, I was away Friday and today. I will continue to look into it tomorrow. So to summarize: Zero tracking error is not really necessary but I need to have movement without sudden jerks that would excite joint elasticities (they are not simulated, but also for the real experiment later on) and clean periodic excitation. I will try to use your suggestions, look for other mistakes and also look at simulator tuning (I'm not sure though what specifics gazebo has in that regard and also which details you mentioned I could be running into). I think that a controller tracking velocity (with position correction) would be the proper solution but I'm fine with using a pure position controller if I get usable results. I'll come back here once I know a bit more, also if not ;)

iron76, 2016-02-09:
Excellent! Keep us updated :)

kjyv, 2016-02-10:
Ok, update: So what I was doing was to generate trajectories in python and since there are no bindings for idynutils from python, went through a gym module that takes the setpoint values and returns the measured values with yarp bottles. The trajectories are better now after improving the performance and syncing to the GYM run loop (before it used delays, well). I also learned from Alfredo (thanks!) that actually position direct is rarely used for iCub trajectory control (which makes sense since going through the whole yarp message stack does not guarantee exact timing, so sending velocity implicitly is not a good idea). Instead position mode actually includes setRefVelocity which apparently allows setting both position and velocity through the generated trajectories. That way proper timing is not necessary anymore. A proper controller that actually controls position and velocity would be better but I'll try this approach for now.

pattacini, 2016-02-10:
@kjyv I'm afraid you've been quite misguided: `position-direct` is widely used today in the iCub control.

pattacini, 2016-02-10:
In a nutshell, if your task deals with the goal of controlling the robot limbs to perform **point-to-point** movements with only one type of trajectory (minimum-jerk in the joint space), then you can profitably rely on `IPositionControl`.

Conversely, if you need to achieve continuous and smooth movements that never stop in between (and with any kind of template for the trajectories), you ought to switch to `IPositionDirect` control.

kjyv, 2016-02-10:
Ok, can you give some more details on why IPositionControl is not suited for continuous smooth control? From what I've learned so far is that depending on the reference velocity that is set, the trajectory will be generated accordingly. Would that result in intermittent stops when a trajectory point is reached or would the speed between those points actually be lower because of lower position errors and lower controller output?
Otherwise, I would reopen the ticket and ask for an interface for a position-velocity controller that can be implemented on the control board level. Even if the position direct solution works in many cases, the proper solution is to be able to give velocity or acceleration profiles at the controller level, not through the whole high level stack.

pattacini, 2016-02-10:
Yes, the final result with `IPositionControl` will be intermittent and quite jerky.

You can still use `IVelocityControl` (or `IVelocityControl2`) for your specific task, which will send to the `FW` the velocity set-points that might be varying at each step. Let me enforce here that these velocity profiles need to be provided through the network as well.

Then, what is proper or not proper remains a matter for debate we could run later :smirk:

kjyv, 2016-02-10:
I tried it with position mode and it actually looks a lot smoother. The benefit of sending velocity over the network would be that these values don't have to be sent in fixed time slots, the values are absolute and not implicitly dependent on a time step that might get corrupted by the messaging pipeline. My "proper" was referring to the fact, that the controller is not actually only running on the control board since the controlled velocity is dependent on the messaging pipeline. But sure, it is a matter of debate ;)
If I was using pure velocity control, I guess there would be position drift after some time which is why a position-velocity would be useful.

pattacini, 2016-02-10:
Normally, the velocity control is closed at high level over the position feedback and this prevents any possible drift.

However, since I might be missing peculiar aspects of your setup, and given that you're at IIT, I reckon it could be worth discussing in person.

kjyv, 2016-02-11:
I'm not sure what you mean with "closed at high level over the position feedback".
But sure, I would welcome talking about these things in person, later in the afternoon?

pattacini, 2016-02-11:
Sure, willingly.