robotology/yarp #675: Problem installing with REGEX mode REPLACE, WINDOWS7

gregoire-pointeau, 2016-02-05:
Hi.
I'm trying to compile yarp on a WINDOWS7 machine (which is a bit old.)

I have the following errors at the compilation: 

> CMake Error at conf/AddInstallRPATHSupport.cmake:150 (list):
  list sub-command REMOVE_DUPLICATES requires list to be present.
Call Stack (most recent call first):
  conf/YarpOptions.cmake:294 (add_install_rpath_support)
  CMakeLists.txt:48 (include)

Then:

>CMake Error at conf/YarpInstallationHelpers.cmake:343 (string):
  string sub-command REGEX, mode REPLACE needs at least 6 arguments total to
  command.
Call Stack (most recent call first):
  data/yarpmanager/tests/CMakeLists.txt:5 (yarp_install)

>CMake Error at conf/YarpInstallationHelpers.cmake:343 (string):
  string sub-command REGEX, mode REPLACE needs at least 6 arguments total to
  command.
Call Stack (most recent call first):
  data/yarpmanager/tests/CMakeLists.txt:8 (yarp_install)

>CMake Error at conf/YarpInstallationHelpers.cmake:343 (string):
  string sub-command REGEX, mode REPLACE needs at least 6 arguments total to
  command.
Call Stack (most recent call first):
  data/yarpmanager/tests/CMakeLists.txt:10 (yarp_install)

>CMake Error at conf/YarpInstallationHelpers.cmake:343 (string):
  string sub-command REGEX, mode REPLACE needs at least 6 arguments total to
  command.
Call Stack (most recent call first):
  data/yarpdataplayer/CMakeLists.txt:11 (yarp_install)

>CMake Error at conf/YarpInstallationHelpers.cmake:343 (string):
  string sub-command REGEX, mode REPLACE needs at least 6 arguments total to
  command.
Call Stack (most recent call first):
  data/yarpdataplayer/CMakeLists.txt:12 (yarp_install)

>CMake Error at conf/YarpInstallationHelpers.cmake:343 (string):
  string sub-command REGEX, mode REPLACE needs at least 6 arguments total to
  command.
Call Stack (most recent call first):
  data/yarpdataplayer/CMakeLists.txt:13 (yarp_install)

This kind of errors are new for me. Does anyone knows where it could come from?

Thanks

pattacini, 2016-02-05:
Did you make sure you started off from a clean system, with brand new repositories?

gregoire-pointeau, 2016-02-05:
Yes, this computer has never seen yarp.
I downloaded the sources this afternoon.
But I realised it a 32Bits...

pattacini, 2016-02-05:
Do you have also an up-to-date version of `cmake`?

gregoire-pointeau, 2016-02-05:
3.5.0

gregoire-pointeau, 2016-02-05:
Visual Studio 2015

pattacini, 2016-02-05:
`cmake 3.5.0` is still unofficial in a sense (it's a _release candidate_), even though I don't really think that's the problem. I'm personally sticking to `3.4.3`.

Those errors are specifically related to cmake, hence @drdanz might help you out.

On the other hand, it's very likely you're going to have problems with `Visual Studio 2015`, since the `ace` package we're providing is not compatible with it (just to say one of the issue I experienced while trying out this version). Thereby, either you compile `ace` from source yourself or you go back to `Visual Studio 2013`.

gregoire-pointeau, 2016-02-10:
I downgraded visual and cmake, and it's working.

closed 2016-02-10 by gregoire-pointeau

lornat75, 2016-02-10:
BTW we should be able to provide Visual Studio 2015 binaries soon. In the meanwhile it would be nice if somebody could give a try compiling the code with Visual Studio 2015 to see if we run into problems.

pattacini, 2016-02-10:
@lornat75 yep, as far as I remember I had experienced problems with the `ace` package we provided, which of course is for versions older than `vs2015`.

lornat75, 2016-02-10:
I ran into similar problems with CMake 3.5.0, this PR fixes it #676.

drdanz, 2016-02-11:
Added a few comments to #676.

I think that the VS2015 generator in CMake 3.5 might have some weird issue, because it reaches parts of the code (`conf/AddInstallRPATHSupport.cmake:150`) where it should not be.

can anyone please try adding this somewhere in the main CMakeLists.txt

```cmake
message("CMAKE_SKIP_RPATH = \"${CMAKE_SKIP_RPATH}\"")
message("CMAKE_SKIP_INSTALL_RPATH = \"${CMAKE_SKIP_INSTALL_RPATH}\"")
message("CMAKE_SKIP_BUILD_RPATH = \"${CMAKE_SKIP_BUILD_RPATH}\"")
```

and give me the output for a few versions of CMake and a few visual studio generators?

drdanz, 2016-02-12:
The patch fixing this is already merged in CMake, it will be in CMake 3.5.0-rc3.