robotology/yarp #677: Error in plugin code running CMake, missing iniruntime file?

lornat75, 2016-02-10:
All plugins give the following error:

CMake Error at conf/YarpInstallationHelpers.cmake:360 (file):
  file COPY cannot find
  "C:/dev/yarp/yarp/src/carriers/human_carrier/human.iniruntime".
Call Stack (most recent call first):
  src/carriers/human_carrier/CMakeLists.txt:7 (yarp_install)

Cmake version 3.5.0-rc1, windows.

barbalberto, 2016-02-11:
The actual name of the required file is human.ini and it is located in the correct folder. 
From CMakeList
```
yarp_install(FILES human.ini
               COMPONENT runtime
               DESTINATION ${YARP_PLUGIN_MANIFESTS_INSTALL_DIR}) 
```

it seems that the word ```COMPONENT``` is skipped and the word ```runtime``` attached to the filename

I tested with cmake version 3.5.0-rc2 on Linux and it happens as well.

It seems to fail here in YarpInstallationHelpers
``` 
set(_oneValueArgs DESTINATION
                    COMPONENT
                    EXPORT
                    LIBRARY_DESTINATION)
  cmake_parse_arguments(_YI "${_options}" "${_oneValueArgs}" "${_multiValueArgs}" "${_what};${_fixedARGN}")
```
Because the ${_YI_COMPONENT}" variable is empty in cmake 3.5 while it is equal to ```runtime``` for cmake 2.8

As a consequences the following replace fails and attach the 'runtime' word to the filename
```
    # Remove COMPONENT argument
    string(REGEX REPLACE ";COMPONENT;${_YI_COMPONENT}" "" _copyARGN "${_copyARGN}")
```

lornat75, 2016-02-11:
ok, the problem seems to be in the last parameter to cmake_parse_arguments. I have commited a fix in  #676 that seems to fix this. Of course we should check that it works with previous versions of cmake. 

BTW: cmake_parse_arguments became built in in cmake 3.5 [1] so maybe we no longer need to include it. 

[1] https://cmake.org/cmake/help/v3.5/command/cmake_parse_arguments.html#command:cmake_parse_arguments

drdanz, 2016-02-11:
> BTW: cmake_parse_arguments became built in in cmake 3.5 [1] so maybe we no longer need to include it.

Finally! They rejected my patch for handling empty arguments years ago, promising this... I wonder if empty arguments are handled or not here...
Anyway we still need to include it for compatibility with cmake 2.8.9. I will have to check that, but I believe that the module is now just a placeholder so that the old projects will not fail.

I will check the patch tomorrow.

drdanz, 2016-02-12:
This is definitely a regression in CMake v3.5.0-rc2. From (at least) CMake 2.8.9 to CMake 3.4, this `CMakeLists.txt`

```cmake
  cmake_minimum_required(VERSION 2.8.9)
  project(bar)

  include(CMakeParseArguments)

  macro(foo)
    set(_options )
    set(_oneValueArgs FOO)
    set(_multiValueArgs )
    cmake_parse_arguments(_FOO1 "${_options}"
                                "${_oneValueArgs}"
                                "${_multiValueArgs}"
                                "${ARGN}")
    cmake_parse_arguments(_FOO2 "${_options}"
                                "${_oneValueArgs}"
                                "${_multiValueArgs}"
                                ${ARGN})

    message("_FOO1_FOO=${_FOO1_FOO}")
    message("_FOO2_FOO=${_FOO2_FOO}")

  endmacro()

  foo(FOO foo)
```

would produce this output:
```
  _FOO1_FOO=foo
  _FOO2_FOO=foo
```

In CMake v3.5.0-rc2 the output is the following:
```
  _FOO1_FOO=
  _FOO2_FOO=foo
```

This means that the last argument is no longer handled in the same way as it was before.

I just wrote an email to the developers mailing list, hoping they will fix it before the final release

drdanz, 2016-02-12:
http://thread.gmane.org/gmane.comp.programming.tools.cmake.devel/15781

drdanz, 2016-02-12:
The patch fixing this is already merged in CMake, it will be in CMake 3.5.0-rc3.
Closing this.

closed 2016-02-12 by drdanz