robotology/yarp #678: yarp guis fail to load qt libraries (Windows)

lornat75, 2016-02-11:
After installing qt manually or from binaries I got the following error when running all qt dependent guis in YARP:

Application failed to start because platform plugin "windows".

This problem happened even if qt libraries were in the PATH. I have found out that CMake (at least 3.5.0) is shipped with its own qt dlls. If you add CMake to the PATH during installation, these library will clash with subsequent installation.

This is the well known Dependency hell (https://en.wikipedia.org/wiki/Dependency_hell). Workaround:

- Remove CMake from your PATH, you do not actually need it unless you work at the command line
- Move your qt installation before CMake in your PATH so that qt's dlls take precedence. CMake seems to be happy, provided you use a compatible version.

kt10aan, 2017-05-04:
I have the same problem, despite not having CMake's PATH set during its installation.

I have to setup a Win10 machine with YARP. The only GUI we need locally is Yarpview.

I am using CMake 3.8.1, Qt 5.8.0 (binaries) and VS 2015. The machine is pretty much in a clean state, as I immediately installed YARP+dependencies (manually) after the Win10 installation.

YARP compiles fine (Release and Debug). The GUIs (Debug version) also work fine.

However, when I run the GUIs (Release version), they produce the same error:

``Application failed to start because platform plugin "windows".``

Copying the plugins/platforms folder to YARP_DIR/bin folder solves that problem and Yarpmanager (Release version) works fine. 

However, Yarpview still does not work, as it obviously cannot find the qml QtYarpViewPlugin.dll.

Here is my path variable:

[PATH](https://1drv.ms/i/s!As2v7hD8lKDjgWSbwjuJgnFv-BcH)

Is it still a PATH issue? Could it be solved by the CMake script copying the needed dlls during installation?
@lornat75, could you list your own variables that you have on your machine?

For now, I get by by using the Debug version of Yarpview, but it is highly inconvenient as I have to explain to other users of the machine that they have to run Yarpview from another directory.

kt10aan, 2017-05-04:
OK, I did find the issue that was causing the trouble. The machine has also MikTeX installed and the MikTex GUI is using Qt dlls, that were interfering with the YARP ones. Removing the MikTeX path solved the issue and all the YARP Release GUIs work properly.

However, its less than an ideal situation. Is it possible to have in the future a YARP installation that copies the necessary dlls in the Release version folder so there is no need for having the QT libs in the path anymore?

drdanz, 2017-05-12:
@kt10aan That would be ideal, but I don't think we have enough man power to take care of that at the moment... There are some CMake modules that do exactly this though.

traversaro, 2017-05-12:
Related CMake module:
* https://cmake.org/cmake/help/v3.8/module/GetPrerequisites.html
* https://github.com/Kitware/CMake/blob/v3.8.1/Modules/GetPrerequisites.cmake

Related issue: 
* https://github.com/robotology/how-to-export-cpp-library/issues/4

anatanevska, 2017-07-06:
Also one possible issue with this can be if you have PyQT or Anaconda in PATH. 

I was getting the same issue (so the error message specifying "Application failed to start because platform plugin "windows"." and then also suggesting windows as a platform), regardless of whether I installed from binaries or source, so I tried just temporarily renaming the Anaconda folder (since i had several Python environments there, all of them with PyQt), and the issue was resolved.

I suppose also reordering the path so that Qt comes before PyQt or Anaconda will work, but for me this approach was the simplest.

kt10aan, 2017-07-06:
@anatanevska, by renaming the anaconda folder, you are essentially invalidating its path. By "temporarily", I assume you mean that you rename the folder every time you want to run the qt guis. Of course it works, but I don't thing it is the simplest approach, especially if you run the Yarp guis often.

anatanevska, 2017-07-06:
@kt10aan  I was just suggesting this as a temporary "workaround" solution, since I didn't see anyone reporting the clash between Anaconda and qt. Of course that it's not a practical solution in the least, but as temporary one it can do. 
Additionally, I saw that for you the issue was resolved by moving the MikTex path from the system to the user path var, which wasn't the case for me (the Anaconda path was already in the user var), so I just added here what worked for me, in case anyone else experienced the same issue.
I absolutely agree that there needs to be a new YARP installation that would cover this.

kt10aan, 2017-07-06:
In general, many programs that use qt5 and get installed come first in the path and cause the failure in Yarp guis. But I am curious, if anaconda was in your user path, then it must have been something *before* that in your user path. Could you post your env variables output? 

Of course, all of these solutions are, to one degree or another, just workarounds. Personally, I only use Yarp on Windows on only one dedicated machine (for VR stuff), so that's not much of an issue for me. But yes, time and resources permitting, a better solution should be preferable

Quiescitus, 2019-03-19:
I had a similar issue of having Anaconda in PATH. Another workaround for this is to run yarpmanager in a terminal opened with a different user account. This prevents local Anaconda paths from interfering with global YARP paths.

closed 2019-03-19 by drdanz