robotology/yarp #679: yarpmotorgui-gt is crashing gazebo

kjyv, 2016-02-12:
Starting yarpmotorgui-qt I enter bigman and select all parts, continuing immediately crashes gazebo.
Part of the output:
[DEBUG]Initializing interfaces...
[DEBUG]Opening interfaces...
[INFO]head iencs->getEncoders() ok!
[ERROR]Error while getting velocity limits, part  head  joint  0

Additionally, after this, yarpmotorgui cannot be quit with ctrl-c (no gui visible), it has to be killed with -9 because it intercepts normal SIGTERMs.
yarpmotorgui-gtk does not have this problem btw.

Gazebo output:
(...)
yarp: Receiving input from /bigman/yarpmotorgui0/head/rpc:o to /bigman/head/rpc:i using tcp
yarp: Receiving input from /bigman/yarpmotorgui0/head/command:o to /bigman/head/command:i using udp
yarp: Sending output from /bigman/head/state:o to /bigman/yarpmotorgui0/head/state:i using udp
yarp: Sending output from /bigman/head/stateExt:o to /bigman/yarpmotorgui0/head/stateExt:i using udp
[FATAL]getJointType() not implemented on your device, cannot proceed further. Please report the problem on yarp issue tracker
/home/stefan/Desktop/robotology-superbuild/build/install/lib/../lib/libYARP_OS.so.1(_ZNK4yarp2os3Log5fatalEPKcz+0x160) [0x7fd78a166ea0]
/home/stefan/Desktop/robotology-superbuild/build/install/lib/libgazebo_yarp_controlboard.so(_ZN4yarp3dev9IAxisInfo12getJointTypeEiRNS0_13JointTypeEnumE+0x4b) [0x7fd7245c727f]
/home/stefan/Desktop/robotology-superbuild/build/install/lib/../lib/libYARP_dev.so.1(+0xeaa2a) [0x7fd74ad67a2a]
/home/stefan/Desktop/robotology-superbuild/build/install/lib/../lib/libYARP_dev.so.1(_ZN4yarp3dev15DeviceResponder4readERNS_2os16ConnectionReaderE+0x58) [0x7fd74acd3938]
/home/stefan/Desktop/robotology-superbuild/build/install/lib/../lib/libYARP_OS.so.1(_ZN4yarp2os4impl20PortReaderBufferBase4readERNS0_16ConnectionReaderE+0x74) [0x7fd78a1cfca4]
/home/stefan/Desktop/robotology-superbuild/build/install/lib/../lib/libYARP_OS.so.1(+0x103b26) [0x7fd78a1ceb26]
/home/stefan/Desktop/robotology-superbuild/build/install/lib/../lib/libYARP_OS.so.1(_ZN4yarp2os4impl8PortCore9readBlockERNS0_16ConnectionReaderEPvPNS0_12OutputStreamE+0x9e) [0x7fd78a1b04ee]
/home/stefan/Desktop/robotology-superbuild/build/install/lib/../lib/libYARP_OS.so.1(+0xf27dc) [0x7fd78a1bd7dc]
/home/stefan/Desktop/robotology-superbuild/build/install/lib/../lib/libYARP_OS.so.1(+0x1627cc) [0x7fd78a22d7cc]
/usr/lib/libACE-6.0.3.so(_ZN21ACE_OS_Thread_Adapter6invokeEv+0x86) [0x7fd789c3ab16]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x8182) [0x7fd7a695e182]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d) [0x7fd7a4ebd47d]
pure virtual method called
pure virtual method called
terminate called without an active exception
terminate called recursively

traversaro, 2016-02-12:
~~~
[FATAL]getJointType() not implemented on your device, cannot proceed further. Please report the problem on yarp issue tracker
~~~

It seems that you are using a relativly new commit of yarp with an old version of `gazebo-yarp-plugins` (`getJointType()` was introduced recently). 

I checked walkman superbuild tags, and apparently in walkman superbuild ( https://gitlab.robotology.eu/walkman/walkman/blob/master/cmake/ProjectsTags.cmake )  you are tracking the master branch of both the repos.

Can you try to update your superbuild (`make update-all && make`) and try again?

traversaro, 2016-02-12:
I just talked with @liesrock and apparently the official superbuild for the walkman project is now https://github.com/robotology-playground/robotology-superbuild . 
The version of YARP and gazebo-yarp-plugins are rather old then. I guess you updated the YARP version to develop your new control interfaces, you can also update the gazebo-yarp-plugins by changing this line : https://github.com/robotology-playground/robotology-superbuild/blob/master/cmake/ProjectsTags.cmake#L7 to point to a recent commit (anything more recent then https://github.com/robotology/gazebo-yarp-plugins/commit/ea65504d6cbbba6bee7dccdecd39b47c0e1ba65a should be ok).

kjyv, 2016-02-12:
Sorry, I should have given more of that info initially. You are right in your assumptions, YARP is updated, GazeboYARPPlugins was not. The problem now is that gazebo 4.1 from the superbuild does not have SetVelocityLimit in the joints api yet. I'm not sure when it was introduced but I guess it could be necessary to update gazebo then....

closed 2016-02-12 by kjyv

traversaro, 2016-02-12:
Yes, I talked with @liesrock regarding deprecating Gazebo <= 4 (see https://github.com/robotology/gazebo-yarp-plugins/issues/226) and he told me that everyone in the walkman/robotology superbuild should use Gazebo 5 (or newer).

liesrock, 2016-02-12:
Probably you had a problem with a previous version of Gazebo already installed: I will force the robotology-superbuild get_dependencies.sh to clean up previous Gazebo version in the system.

kjyv, 2016-02-12:
I'm pretty sure it installed gazebo4 on its own, but I've updated to 5 without problems.

liesrock, 2016-02-12:
Ok, thanks for the report. I will check it.