robotology/yarp #684: Cannot run YarpDataDumper on Localhost

ashrafzia, 2016-02-15:
Hi,

I cannot properly start yarpdatadumper from GUI. It gives the following error:

`[ERR] (dataDumper_for_robot) cannot run yarpdatadumper on localhost : `
 
When i try opening on terminal, i have the following error message:

```
yarpdatadumper --name /data/rightArm
||| configuring
||| no policy found

[ERROR]unable to open file:  ./data/rightArm/info.log 
yarp: A thread failed to start with error code: 0
Module failed to open

```
I don't get when i try to refresh the **yarpdatadumper_recording_example** list, some **datadumpers** are in running state while most of them are stopped. I am only interested in the rightArm and head. How do i troubleshoot them?

Thank you.

pattacini, 2016-02-15:
Hi @ashrafzia 

When you tried to launch `yarpdatadumper` from within a terminal session you got an error saying that the system was unable to open a file.

Do your user rights allow for creating a file in that specific location on the drive?

ashrafzia, 2016-02-15:
Ah, Thanks.

It was privileges issue. Its working now. But there's no data in the log file. I run my program and moved the rightArm to a new location in the cartesian space followed by the eyes. When i terminate my program using Ctrl-C and check data.log and info.log, there's no data.

pattacini, 2016-02-15:
It's likely you forgot to establish connections among the robot and the pool of your dumpers :smirk:.

ashrafzia, 2016-02-15:
hmm......I doubt something.

Since i am using a while(1) loop in main.cpp, and when i have moved the arm to a particular point in the space and looked at that point, on my terminal i get the following messages:

```
yarp: Removing output from /client/right_arm/command:o to /icubSim/cartesianController/right_arm/command:i
yarp: Removing input from /icubSim/cartesianController/right_arm/state:o to /client/right_arm/state:i
yarp: Removing input from /icubSim/cartesianController/right_arm/events:o to /client/right_arm/events:i
yarp: Removing output from /client/right_arm/rpc:o to /icubSim/cartesianController/right_arm/rpc:i

```
main.cpp

```
#include "kinArm.h"
int main(int argc, char *argv[]){
	Network yarp;			   
	kinArm arm;
    string menuSelection;
    string selectedElements, selectedElement1;
    	selectedElement1 = arm.menu();
		cout<<"\nFirst Element: "<<selectedElement1<<endl;
		arm.initPorts( selectedElement1 );
		arm.deviceDriver();
	while(1){
		arm.initCub();
	}
    return 0;
}
```

kinArm.cpp
```
#include "kinArm.h"
kinArm::kinArm(){
    nj = 0;
    done = false;
    times = 0;
    count = 50;
    which_joint = 0;
    lower = 0;
    upper = 0;
    command.resize(nj);
    f_nextElement = true;
    startup_context_id1 = 0;
    startup_context_gaze_id = 0;
    fp.resize(3);
}

/* -----------------------MENU------------------------------------------------ */

string kinArm::menu( void ){            
    string element_name;
        cout<<"\t========= MENU ========="<<endl;
        cout<<"\t0. Head"<<endl;
        cout<<"\t1. Torso"<<endl;
        cout<<"\t2. Left Arm"<<endl;
        cout<<"\t3. Right Arm"<<endl;
        cout<<"\t4. Left Leg"<<endl;
        cout<<"\t5. Right Leg"<<endl;
        cout<<"\t6. Exit"<<endl;
        cout<<"\t========= MENU ========="<<endl;
        cout<<"Please Select Element No: ";
        cin>>which_element;
        switch( which_element ) {
            case 0:
                element_name = "head";
            break;

            case 1:
[... 135 lines cut ...]
void kinArm::eyesTarget( Vector a ) {   
        fp[0]=a[0];                                 // x-component [m]
        fp[1]=a[1];                                 // y-component [m]
        fp[2]=a[2];                                 // z-component [m]        
        igaze->lookAtFixationPoint(fp);                 // move the gaze to the desired fixation point
        igaze->waitMotionDone();                        // wait until the operation is done
        printEyesStatus();
}

void kinArm::printEyesStatus( void ){        
        // we get the current fixation point in the operational space
        cout<<"\nPrinting Eyes Status......"<<endl;
        Vector x;
        od = o0;
        igaze->getFixationPoint(x);                     // retrieve the current fixation point
        cout<<"+++++++++"<<endl;
        cout<<"fp         [m] = "<<fp.toString(3,3).c_str()<<endl;
        cout<<"x          [m] = "<<x.toString(3,3).c_str()<<endl;       
        cout<<"norm(fp-x) [m] = "<<norm(fp-x)<<endl;    // return a measure of the displacement error
        cout<<"------------"<<endl<<endl;
        igaze->getHeadPose(x,od);
        cout<<"\nHead Pose: x[m] = "<<x.toString(3,3).c_str()<<endl<<endl;       

}
```

pattacini, 2016-02-19:
Any news on that?

closed 2016-02-26 by pattacini