robotology/yarp #690: OpenNI2 device and possible migration to RGBDWrapper

kt10aan, 2016-02-23:
Here is @barbalberto's post from issue #683 :

Ok, I see there is there is good will for improvement and I like it :+1: so here is the 'long answer', I hope you guys can read it all and provide feedback.

My interest in the OpenNI device came from issue #643 I'm in charge to work on.

I examined all drivers related to kinect-like devices (in IIT we need one in a new robot) and I recently implemented the same in the Gazebo simulator (https://github.com/robotology/gazebo-yarp-plugins/issues/100) so we can work on simulator and then go on the real robot.

What I found out is that there is no real 'standard' in yarp for this kind of devices, and the situation is even worse for pointclouds. So I'd like to make some order in this.

I found out there are 4 kinect-like yarp drivers but all of them lack of some key features or have some dis-advantage.
Please think of the following points as ways to improve the yarp-ish software related to this awesome depth cameras devices.
My observations are:
**A)** Driver in yarp are meant to be splitted into 2 different part (modules/class/object):
  - Proper driver: this part reads the data from hardware (no yarp port!)
  - Wrapper: this part uses an interface to read data from the former one and it send the data on the network through yarp ports.
The reason is to re-use the same wrapper with different hw-devices, i.e. we will have the same ports and same data, same rpc calls regardless of using kinect, xtion or other devices (ideally also a streovision sistem with 2 cameras and disparity) through abstraction layer. Obviously abstraction layer has to be comprehensive to cover small differencies that can arise between different hw producers.

This is what is done for motioncontrol, so we use the same controlBoardWrapper2 over CAN devices, ETH devices, simulators (iCub, gazebo, robotran), ethercat device and so on. This allows code to work on different robots by abstracting hw level.  
All kinect-like yarp drivers I saw, do the 2 things (driver and wrappper) together in a single module and cannot be separated. This means, for example, there was no 'pure wrapper' I could use with the simulators, therefore I had to write a yet another one.

**B)** As follow from lack of an unified wrapper, each device opens it own ports and send data using is own format: 
- OpenNI2 uses one port for ```rgb``` and one port for ```imageOf<PixelMono16>```  which is unsigned int
- robotology/kinect-wrapper uses one port for ```rgb``` and one port for ```imageOf<PixelMono16>```  which is unsigned
- yarp/modules/kinect/OpenNi(1) uses  ```ImageOf<PixelInt>``` which is ```int32``` with *sign*
- yarp/modules/kinect/freenect uses ```PortablePair<yarp::sig::ImageOf<yarp::sig::PixelMono16>``` this is a pair, so different type

This make the devices incompatible between each other *even if they basically provide the same data*, therefore software made to work with one driver cannot work with another one. The usage of an unified wrapper/interface will make possible to test the same software with different sensors or change the sensor for example from kinect to xtion effortlessly.

**C)** Each device is a bit customized in the way it works, for example both OpenNi2 and kinect-wrapper has a port for skeleton, but their definition of skeleton is not the same while the other 2 devices do not provide skeleton.
This again goes against the re-usage of the code.
Following comment from @kt10aan
> The philosophy behind the driver is exactly that, of a driver: exposing the functionality and providing an interface to what is available on the physical device and just streaming the data 

to which I agree, I think that since skeleton or pointclouds are not data provided by the sensor, but a software elaboration on top, this feature should be extracted and implemented on a separated module. Again, using this approach will allow user to test the same algorithm (library) for skeleton extraction on different hw or to use different skeleton extraction algorithms (libraries) on the same hw. Those modules could be run on a remote machine more powerful. If I have (and I will) have the sensor installed on board of the robot, maybe compiting skeleton extraction or point cloud generation on board could be too heavy.

Another advantage is to remove dependencies for compilation. I'm not able to install Nite2 on my machine and, at the end of the day, I don't need it either but I got compilation issues because it is required.

So, to cut a long story short, as discussed internally with @lornat75, @drdanz and others, the idea in the long run is to have a 'infrastructure' similar to the one we have currently for motor control with:
- 1 wrapper/client for all RGBD devices
- many driver as required: kinect1 - kinect2 - OpenNI(1) - OpenNi2 - ...  sending data ```rgb & depth```
- One (or more) module doing skeleton extraction from images  (hopefully with a compatible skeleton definition)
- One (or more) module generating pointclouds from images

What have been done so far:
- Interface for depth sensors: https://github.com/robotology/yarp/blob/master/src/libYARP_dev/include/yarp/dev/IDepthSensor.h
- Interface for RGBD sensor: https://github.com/robotology/yarp/blob/master/src/libYARP_dev/include/yarp/dev/IRGBDSensor.h
  This one basically inherits from FrameGrabber and Depth sensor adding a method to get data in synch

- RGBD client/wrapper pair: https://github.com/robotology/yarp/tree/master/src/libYARP_dev/src/modules/RGBDSensorClient
https://github.com/robotology/yarp/tree/master/src/libYARP_dev/src/modules/RGBDSensorWrapper
   Those devices are meant to work together to send/receive data and set/get configuration thorugh RPC calls
- Gazebo plugin for depth camera sensor
- Module converting images into ROS pointcloud. It'll create a yarp point cloud too, as soon as the yarp-ish point cloud type is finalized and implemented. ROS compatibility will still be granted as well. (This is not in yarp repo yet.)

The definition of the interface is tentative, so features can be changed or added. For example mirroring is not present in the interface right now. We could add it so a sw can remotely know how is the mirroring set and comply accordingly. 

Anyway, right now the client/server just exchange the image streams, the *RPC calls are not yet implemented*.
Probably first step will be to create a new OpenNi2 compatible device to be used with the RGBDWrapper and then help migration toward it.

What do you guys think? 
Since it is a piece of work, con you spare some time to help in this process?

I do believe the result will make things more usable clear, improve the reusage of code and share of result and modules.

kt10aan, 2016-02-23:
Soon, let me give an detailed reply then :)

- First of all, regarding the four available softwares for depth sensors: The OpenNI1 and the freenect based modules should be considered obsolete as OpenNI1 is abandoned and libfreenect was a first attempt and works only with the Kinect 1.Also, libfreenect is more of a driver to let Kinect work with Linux and OSX than a full depth sensor framework. The remaining two, the OpenNI2 device and the Kinect wrapper are very different in philosophy, as you can see by a quick examination of their documentation pages and the available parameters.

For the rest of the reply, I will ignore the OpenNI1 and the libfreenect based devices, in order to simplify things.

- Some explanations, because the terminology that applies to the depth sensors situation is not that simple as "driver-implementation".
We have hardware devices: Kinect1, Kinect2, Asus Xtion etc.
We also have the SDKs (for lack of a better work) for these devices: Kinect SDK, OpenNI2

Kinect1 and Kinect2 have their own (not open-source), Windows-only SDK.

Asus Xtion and ALL Primesense-compatible devices (Primesense was the company initially behind OpenNI, before being sold to Apple), use OpenNI1 and OpenNI2. OpenNI1 and OpenNI2 are open-source. After Primesense stopped the development of OpenNI1 and 2, another company picked up OpenNI2 and leads the development, while also in the process of producing more sensors (among others, some quite small ones for use with the iPhone...)

Also, OpenNI2 is multi-platform (Windows, Linux, OS X)

BUT kinect1 and kinect2 can also work with OpenNI2 through a driver (libfreenect and libfreenect2) (not to be confused with the YARP module freenect)

So, to summarise:
OpenNI2: multiplatform, opensource, works with multiple devices
Kinect SDK: Windows only, closed source, works only with Kinect devices.

In my mind, that is an easy choice.

- Regarding the dependencies: The NiTE2 dependency for the OpenNI2 YARP driver is supposed to become optional. Its not yet because, time!

- Regarding the non-uniform outputs of existing devices/modules: Yes, it is a very good idea to standardise the depth and RGB output. However, the Skeleton output cannot be standardised (not easily anyway) as each SDK is using a different number of joints.

I would be glad to convert the OpenNI2 YARP driver to the new format/wrapper. It wouldn't take so much work I assume and it could be ready for the next YARP version. It would be great to standardise as much as possible. But also, some discussions have to take place in order to decide some design/standardization issues that are not yet addressed in the iDepthSensor and IRGBDSensor interfaces :)

barbalberto, 2016-02-23:
Thanks a lot @kt10aan for your clarification!

So if I understood correctly you suggest to:
 - drop support for OpenNI1 
 - drop support for freenect1 --> does this means drop support for Kinect1 or freenect2 supports both kinect 1 and kinect2?
 - support PrimeSense compatible devices through OpenNI2 on Win/Linux/OSX
 - support Kinect2 using freenect2 on Linux
 - support Kinect2 using Windows SDK on Win 

> Regarding the dependencies: The NiTE2 dependency for the OpenNI2 YARP driver is supposed to become optional. Its not yet because, time!

I made it on my branch https://github.com/robotology/yarp/tree/openNI2_device_clean_wip, you can check it out. If it is fine with you guys, I can merge it into master. 
Basically the CMake checks for Nite2 and, if it is found, the skeleton extraction is compiled, otherwise it isn't.
I suggest to merge your changes about the mirroring, then I'll update the branch.

> It would be great to standardise as much as possible. But also, some additional discussions (including @lornat75 and @drdanz I assume) have to take place in order to decide some design/standardization issues that are not yet addressed in the iDepthSensor and IRGBDSensor interfaces :)

Good, the issue #643 is there for discussion. I think all comment related to the standard can be posted there and we will follow the implementation of the standard for the OpenNI2 device here.
For example mirroring is not supported in the interface right know, we can add it. _If you have other suggestions please this is the best moment to state them so we can add it in the new interface/devices._

> However, the Skeleton output cannot be standardised (not easily anyway) as each SDK is using a different number of joints.

I agree that part is more difficult to standardize. I never worked with skeleton so it is not clear to me, however ideally we should choose a yarp-ish definition of what a skeleton is, flexible enough to be able to be converted to and from skeleton defined by other SDK. Something like we do for images, they are compatible with opencv, with ROS but does not exactly match any of them. This will give us more freedom ... anyway it is not important right now. Let's do one step at the time.

barbalberto, 2016-02-23:
Another question:
what information NiTE2 need to operate? Is it enough to have rgb and depth images or it needs something else?
Does it requires to have access to the HW device somehow?

kt10aan, 2016-02-23:
Strictly speaking, NiTE2 needs only the depth stream to do the skeleton tracking

In the openni2 driver, the access to HW is needed if you are using a live stream (obviously). But the HW is not needed if you feed to it an .oni file.

jgvictores, 2016-02-23:
@barbalberto I tried https://github.com/robotology/yarp/tree/openNI2_device_clean_wip with NiTE, and skeleton tracking is not available. It actually does not reach the step where it looks for s.dat; this is, nor "Could not find data file ./NiTE2/s.dat" nor "User tracker started... Start moving around to get detected... (PSI pose may be required for skeleton calibration, depending on the configuration)" are reached. ((BTW rgb and depth work))
If you do not have NiTE, I store the working copy https://sourceforge.net/projects/roboticslab/files/External/nite/NiTE-Linux-x64-2.2.tar.bz2 got from http://ilab.usc.edu/packages/forall/ubuntu-13.10/. On which OS are you?

barbalberto, 2016-02-23:
Thanks for pointing me to the NiTE library, so I could test the code with and without.
There was an error in the CMake. I just pushed an update, can you try again please?

barbalberto, 2016-02-26:
@jgvictores Have you tried new branch to see if it works ok now?

jgvictores, 2016-02-26:
@barbalberto Sorry, been 100% with IROS2016!! @@ Have meeting now, but will check right after!

kt10aan, 2016-02-26:
Busy with IROS2016 here too! Please remember that most of the testing needed for the OpenNI2 driver (speaking from experience) is if all the combinations of the parameters work with each change. This is where most of the functional errors appear after each addition/change and are the hardest to catch without extensive testing of all possible combos of the parameters and with more than one sensor.
The structure of OpenNI2SkeletonTracker.cpp is quite complex at points.

kt10aan, 2016-02-26:
@barbalberto, also my changes for the mirroring will have to wait until 1st of March to be fully tested and merged, sorry, it is going to be a crazy weekend regarding work :(

kt10aan, 2016-02-26:
@barbalberto,

also, to answer your question above where you mentioned, as you seem to confuse the YARP freenect device with the libfreenect project.

> So if I understood correctly you suggest to:
>
>drop support for OpenNI1
I do suggest to drop support for OpenNI1. It is not maintained for years now. The legacy YARP driver could be kept around, if somebody needs it.

>drop support for freenect1 --> does this means drop support for Kinect1 or freenect2 supports both >kinect 1 and kinect2?
I do suggest to drop support for the _YARP Device freenect_ . Do not confuse the YARP freenect device, the linux library libfreenect which is primarily a hardware driver for Kinect1, the linux/OS X library libfreenect2, which is primarily a hardware driver for Kinect 2.
Dropping the totally obsolete _YARP Device freenect_, does NOT drop support for Kinect 1. OpenNI2 supports Kinect1 fine through the libfreenect (the linux/OS X library, not the YARP module!). The current YARP OpenNI2 device supports Kinect (see the documentation at the wiki).

>support PrimeSense compatible devices through OpenNI2 on Win/Linux/OSX
Affirmative on the Primesense devices

>support Kinect2 using freenect2 on Linux
No, I suggest to support Kinect2 using OpenNI2 and the (linux/OS X library) libfreenect2. 

>support Kinect2 using Windows SDK on Win
If someone wants to make a Windows specific YARP-device, nobody stops him. However, Kinect1 and Kinect2 can also be used on Windows with OpenNI2 (the user has to install the corresponding Windows Kinect SDK for the hardware driver, but the rest of the handling is being done by OpenNI2)

I hope the above clarifies things a bit :)

kt10aan, 2016-02-26:
@all, a summary for what the converted to the new format OpenNI2 YARP driver will support and on what OS.
- Primesense compatible devices on Windows/Linux/OS X
- Kinect 1 on Windows/Linux/OS X (it will additionally need the installation of libfreenect on linux/OS X, or Windows Kinect SDK on Windows)
- Kinect 2 on Windows/Linux/OS X (it will additionally need the installation of libfreenect2 on linux/OSX, or Windows Kinect2 SDK on Windows)

Right now, as it is, the OpenNI2 YARP driver supports:
- Primesense compatible devices on Linux/OS X (it should support Windows too but I never wanted to go through the pain of testing it:D)
- Kinect 1 on Linux/OS X (it additionally needs the installation of libfreenect, as mentioned in the documentation. Again, the same as above apply for Windows support
- Kinect 2 is not supported yet, mostly because the libfreenect2 project is quite new and needs a bit of maturing and stability (which will happen quite soon judging by their progress)

barbalberto, 2016-02-26:
@kt10aan thanks a lot for your precious information.

About noUserTracking, I didn't noticed it. I just pushed a commit to set the default to false if NiTE2 is not found and true if NiTE2 is found.

It is not so urgent to be merged, so stay focused on IROS! 
I can do some simple test myself, just to see if the changes does not create weird behaviours.Can you guys point me to some module using the skeleton?

I know the difference between libfreenect and yarp driver, my confusion cames from what's the role of OpenNI2. At the beginning I thought it was a library on its own, similar to libfreenect but specialized to work with a different type of hw.
From what you say instead it seems that OpenNI2 is actually a framework working **on top** of libfreenect (both 1 or 2) or Win SDK. Is it right?

jgvictores, 2016-02-26:
@barbalberto I tried the `openNI2_device_clean_wip` again; got skeletons back :smile: Sorry, I'm not aware of modules that use them though. BTW, could you add support for `rgbMirror` and `depthMirror` as in https://github.com/kt10aan/yarp/blob/3ad6c8175882cb28495ca25c37b007ffdc3d15e4/src/modules/openni2/OpenNI2DeviceDriverServer.cpp? ...or maybe wait for clarifying defaults and error treatment at #683.

kt10aan, 2016-02-26:
@jgvictores, I will push from my fork the mirroring changes and then  @barbalberto can push his changes (but it would be nice if i also had time to test his branch :) ).

@barbalberto, the confusion about what OpenNI2 is, comes from the fact that it is two things in one:

- A hardware driver for Primesense-compatible devices, like Asus Xtion. The company Primesense (that produced OpenNI1 and OpenNI2) was also making RGBD sensors. In a nice and smart move, they were allowing other companies to produce Primesense-compatible devices (like Asus Xtion)
- A framework on top of the Primesense-compatible hardware driver for getting the data out of the sensor and preprocess them, before being passed to some middleware (e.g. NiTE for skeleton tracking).

Kinect devices could not be used, at least at the beginning, with OpenNI, as a hardware driver was needed. For Windows, the hardware driver part of the Kinect SDK could be used and then OpenNI would then take over. For Linux/OS X, where there was not Kinect SDK, the hardware driver part was solved by the libfreenect project. It enables OpenNI to work with Kinect devices.

Nice commit regarding the userTracking :) I will test exhaustively after Tuesday.

kt10aan, 2016-02-28:
Also, it is worth noting that Occipital, the company that took over development of OpenNI2, has released a very very small but fully capable sensor, that (apart from working with iOS), also works with OpenNI2 and virtually on any platform (including small ARM boards).

https://store.structure.io/store

We are going to order one of these sensors. It works with OpenNI2 out of the box :)

gsaponaro, 2016-03-03:
Another company selling RGB-D hardware sensors that support OpenNI2 (plus their own SDK) is Orbbec:

https://orbbec3d.com/

Haven't tried those personally yet, but they seem to have good specs.

barbalberto, 2016-03-09:
We have one from https://store.structure.io/store, and I can confirm it works with openni2 device.
It has only depth sensor and not rgb one. The driver tells it cannot read rgb and goes onward.

I have not done extensive test on that but I was told the xtion provides better measurements.

I see the first merge has been done, can I proceed to merge the patch about NiTE2 exclusion when not found?

kt10aan, 2016-03-09:
@barbalberto, could you please wait one-two days (e.g., until Friday afternoon?) Things got insanely busy here :)
Good news regarding the Structure sensor. 

I am also preparing a detailed post regarding the RGBD interface and questions/suggestions I have about it :) As soon as we can settle on the RGBD interface I will start the porting of the OpenNI2 driver.

barbalberto, 2016-03-10:
:+1: Let's set it for monday 

When you are to start porting tell me, we can chat a bit on details, like if new things are required in the interface and about skeleton

kt10aan, 2016-03-14:
@barbalberto, no problems  found here with the NiTE2 exclusion patch. :)

kt10aan, 2016-04-19:
After vacations, I am back and ready for some RGBD action (that sounded wrong!).

So, @barbalberto, here are some comments/questions regarding the IDepthSensor/IRGBDSensor wrappers. Keep in mind that my view on the following is mostly from the aspect of OpenNI2 (since it is the most widely used and open-source) but also from the aspect of the the depth sensors and APIs I have worked with up to now.

1. **`getMeasurementData`**: 
isn't it the same as the depth frame? RGBD sensors output two channels of information: an RGB video stream and a Depth video stream. Of course, no harm in having the method, just that I do not think it will be used with OpenNI2.

2. **`getDistanceRange/setDistanceRange`**: What do you mean by "range" here? Range to what? Because each pixel in the depth frame represents the "range" of that specific spot.
What I have used in the past is a "cutoff" range, where beyond that, depth info is discarded, to reduce noise.
But in that case, should it be named `getCutOffRange` or something similar?

3. **`getHorizontalScanLimits/getVerticalScanLimits`**: I dont think these will be used with OpenNI2 as it is usually fixed by the focal properties of the sensor. No harm in having them though.

4. **`getDataSize/setDataSize`**: Again, no harm in having them but I dont see a practical use for OpenNI2 (see `getMeasurementData` above)

5. **`getScanRate/setScanRate`**: I assume this would correspond to frames per second. Maybe `getFPS` would be a better name?

6. **What about a skeleton representation?** Just as a reminder, NiTE2 has the following data regarding the skeleton 15 Joints: each joint has x,y,z position, position confidence (float), a wxyz quartenion for the rotation and rotation confidence (float)

Since different skeleton tracking middleware have different structures and number of joints, I would suggest to just define a struct or object for a single joint and an array for a skeleton with varying numbers of joints.

Of course, skeleton tracking would be an optional thing, but I do think that a minimal joint/skeleton structure should be present to standardize things in the future.

cc @gsaponaro, @jgvictores

jgvictores, 2016-04-19:
@kt10aan @barbalberto @gsaponaro 
Given (Xtion Pro example):
`Depth video modes available: 0: 320x240, 30 fps; 2: 320x240, 60 fps; 4: 640x480, 30 fps; 6: 160x120, 30 fps`
`RGB video modes available: 0: 320x240, 30 fps; 3: 320x240, 15 fps; 6: 320x240, 60 fps; 9: 640x480, 30 fps; 12: 640x480, 15 fps; 15: 1280x1024, 30 fps; 19: 1280x720, 30 fps; 21: 1280x960, 30 fps`
Regarding setDataSize and setScanRate (BTW, I support setFPS a bit more too):
1. Should there be separate methods for RGB vs D ?
2. What happens in this case where Size and FPS are coupled?

barbalberto, 2016-04-20:
Hi guys, thanks for the comments, I really appreciate.
I'd like to start soon to implement rpc calls currently missing for wrapper/client. 
While looking back at the interface I saw some methods had un-happy names, like dataSize you pointed out.
That one is probably going to change. I'd like to have it as intuitive as possible and my idea was to have it more similar to standard camera methods.

Some of them have names that could sound weird at first look, each have a reason. I'll answer in detail ASAP (we have a demo today)

One question about skeleton:
I see some cases skeleton are tree-like structures https://www.google.it/imgres?imgurl=http%3A%2F%2Fsocial.msdn.microsoft.com%2FForums%2Fgetfile%2F152841&imgrefurl=https%3A%2F%2Fsocial.msdn.microsoft.com%2FForums%2Fen-US%2F24718d31-8010-40c0-b2ed-c5ec82ccdb80%2Frgb-and-depth-frame-alignment-in-c%3Fforum%3Dkinectsdk&docid=txU0FeFpBJBn-M&tbnid=eYhpBG3uMpNO8M%3A&w=644&h=478&safe=off&bih=796&biw=1536&ved=0ahUKEwj31erWw5zMAhVD8RQKHSiYAIIQMwhfKDgwOA&iact=mrc&uact=8 
and in other cases they have closed structures https://www.google.it/imgres?imgurl=http%3A%2F%2Fforum.libcinder.org%2FviewImage.do%3FfileId%3D23286000000621009%26forumGroupId%3D23286000000003001&imgrefurl=https%3A%2F%2Fforum.libcinder.org%2Ftopic%2Fopenni-libs-for-kinect&docid=S6pPw2Q8vB7dSM&tbnid=qkUIfQEByjFoCM%3A&w=640&h=502&safe=off&bih=796&biw=1536&ved=0ahUKEwi2jePgw5zMAhVCaxQKHb6BAY04ZBAzCDAoLTAt&iact=mrc&uact=8

so the definition should be able to support both cases.
Can you show me an example of joint/skelet description? Which information are needed?

Thanks a lot

jgvictores, 2016-04-20:
@barbalberto I agree deciding an API for skeletons is good, but I agree with what you said on #683: `I think that since skeleton or pointclouds are not data provided by the sensor, but a software elaboration on top, this feature should be extracted and implemented on a separated module. Again, using this approach will allow user to test the same algorithm (library) for skeleton extraction on different hw or to use different skeleton extraction algorithms (libraries) on the same hw. Those modules could be run on a remote machine more powerful. If I have (and I will) have the sensor installed on board of the robot, maybe compiting skeleton extraction or point cloud generation on board could be too heavy.`.
So... the skeleton API should be defined at a different level, possibly in a different header file?

-

> I see some cases skeleton are tree-like structures, and in other cases they have closed structures.

Essentially, I see no tree structure. I believe that's pure visualization, just 'connecting the dots' in different ways.

> Can you show me an example of joint/skelet description? Which information are needed?

As for inspiration, in the Windows Kinect SDK 1.0, skeleton data was in a [NUI_SKELETON_FRAME Structure](https://msdn.microsoft.com/en-us/library/nuiskeleton.nui_skeleton_frame.aspx), so you could access data as in: 
`SkeletonFrame.SkeletonData[i].SkeletonPositions[NUI_SKELETON_POSITION_ELBOW_RIGHT].x`
`SkeletonFrame.SkeletonData[i].SkeletonPositions[NUI_SKELETON_POSITION_HAND_RIGHT].y`
Since we're getting a bit more high-level, I find `SkeletonFrame.SkeletonData[i].eSkeletonPositionTrackingState[NUI_SKELETON_POSITION_WRIST_RIGHT]` (to check how **tracking** is performing) also quite interesting.

A glimpse of ye-oldie YARP OpenNI (1/2?) module, with a similar notation (`userSkeleton->skeletonPointsPos[jointIndex][0] = joint.position.position.X;`) adds **orientation** and **confidence** (that I recall was always set to 1.0, so not really useful info) values.

-

As for the API, I see two options (if confidence is only 0/1, it could be merged into trackingStateEnums):
- Getting a bulk of vector-of-vectors (I think this is the case of [IOpenNI2DeviceDriver](http://www.yarp.it/IOpenNI2DeviceDriver_8h_source.html), with functions like:
`virtual bool getSkeletonPosition(const int& userID, yarp::sig::Vector& vectorArray, yarp::sig::Vector& trackingStateEnums, yarp::sig::Vector& confidence) = 0;`
- Getting coordinates and info per-joint-frame, as in:
`virtual bool getSkeletonPosition(const int& userID, const int& jointID, yarp::sig::Vector& jointFrameCoordinates, int& trackingStateEnum, double& confidence) = 0;`

barbalberto, 2016-04-27:
Hi guys, here is the complete reply, sorry if I'm too verbose ...

Ok, let's start from the beginning.

`IRGBD` interface is the sum of the historical `FrameGrabber` interface for rgb camera and the new `IDepth` interface for the depth part.

`yarp::dev::IRGBDSensor:  public IFrameGrabberImage,
                          public IDepthSensor`

When I started designing the iDepth/iRGBD interfaces I wanted to make them the more general as possible. I hoped the iDepth interface could be used by any kind of depth sensors: cameras but also linear scanner lasers and point laser etc... So the names of methods right now are meant to be as generic as possible and not camera-oriented.

Then after working for a bit, I realized it was not such a good approach because too generic becames confusing, so the `IRangefinder2D`interface kicks in for linear laser (and single point laser) so the `iDepthSensor` can be more focused on camera-like sensors.

Probably I'd change the name of the interface into `iDepthFrameSensor` to reflect this approach.
So I'd like to adapt names to reflect this new approach and make them more intuitive to use.

@kt10aan

1 `getMeasurementData`: This was meant to be _the_ method intended to retrieve the depth frame, so this is gonna be used by OpenNi2. Name was simply about `data` to be generic, but now I think calling it `getDepthFrame` or similar would be better.
Note: this is gonna retrieve ONLY depth since it is part of the `iDepth` interface, the `iRGBD::getRGBDFrames` will retrieve both rgb and depth frame.

2 `getDistanceRange/setDistanceRange`: this is _similar_ to the cutoff you talk about. The difference is that this reflects the HW capabilities. For example the Xtion is able to detect objects starting from a minimun distance of 40cm up to 4m. If the object is too close to the sensor you'll get `nan` as a reading, the same if the object is too far from the sensor. This is an HW limitation and we cannot do anything about it, while the cutoff is more a software filtering. So getting this values from the HW is useful for the user to know the sensor should be at a certain minimun distance from the object in order to get valid data. I don't know any device that allows to set this value so maybe we can drop the set. 
I don't like using cutoff here because, although the effect is probably the same, it cames from a different reason imho. Furthermore I'd like more a name that make me think about the data I get (hw capabilities) instead of the data I drop.

3 `getHorizontalScanLimits/getVerticalScanLimits`: usually it is fixed, but it is needed by the algorithm that will create pointclouds from rgbd data so there must be a way to get it. 

4 `getDataSize` this is the size of the depth frame, i.e. the resolution of the frame. Again, using an approach more camera-oriented this can be changed into heigth/width or something else.
As @jgvictores pointed out (thanks, I didn't know) the Xtion allows to set different values of resolution for the depth sensor, so a way to set it is useful. Also I think that it is good to support sensors with different resolution values for rgb and depth, so both interfaces (rgb and depth) shall have their way to get/set those values.

5 `getScanRate`: yes. In the original idea this interface would be used also for lasers to which the idea of frames would not apply so I choose a different name. Now I think changing it to getFPS is better.

@jgvictores

1. Should there be separate methods for RGB vs D ?
yes, there will be 3 methods:
 - `IFrameGrabberImage::getImage(...)` for rgb 
 - `IDepthSensor::getMeasurementData(...)`  for depth -> this is probably going to be renamed `IDepthFrameSensor::getDepthFrame`
 - `IRGBDSensor::getRGBD_Frames(...)`  getting both frames in sych in a single call  (since IRGBD inherits from the other 2, you can access all three of them from the IRGBD interface only with `IRGBDSensor::getMeasurementData and IRGBDSensor::getImage)`

2. What happens in this case where Size and FPS are coupled?
That's up to the HW. Regular rgb cameras sometimes change coupled values togheter when setting one of them. User have to verify the configuration after setting desired values. The interface will not take any action in this regard.

I made a bit of work in the RGBD_rpc branch, in particular I changed dataSize into height/witdth for the iDepthSensor interface, while when using iRGBD interface you get rgbDataHeight/Width and depthDataHeight/Width.  (We can use Frame in place of Data)

https://github.com/robotology/yarp/blob/RGBD_rpcs/src/libYARP_dev/include/yarp/dev/IRGBDSensor.h

About skeleton I'm asking because we are in the topic, but I agree with @jgvictores that skeleton should not be in the device and defined in a different place. We still have to idenity the best place; in the meanwhile I'd like to get an idea of how this type may be.

Maybe we can open another issue to discuss this topic specifically

gsaponaro, 2016-04-27:
Hi all, a few thoughts:

- I agree that the skeleton part can be done later, in a separate class;
- in order to get the RGB frame the existing function is called ```IFrameGrabberImage::getImage```; on the other hand, in order to retrieve the Depth frame the proposed function is ```IDepthSensor::getMeasurementData``` or ```IDepthFrameSensor::getDepthFrame``` - I find it a bit confusing that the word "frame" appears in the first function case but does not appear in the second one for symmetry;
- ```getDataSize``` discussion, related to the fact that sometimes "Size and FPS are coupled". I am not sure if we are talking only about the initialization/startup phase, or re-setting some of the parameters online during driver operation (if it is even possible). It is true that different RGB-D sensors have different **modes** permitted (I call **mode** a combination of width x height x fps). There are even variations between different physical sensors of the same model (due to slightly different hardware variations), for example, talking about the Xtion, [these](https://gist.github.com/gsaponaro/a620985df86cfb82c0ea92d6bbb85320) are the modes that I have access to, while [these](https://gist.github.com/gsaponaro/33adcac19ed0c23c24d451158068992c) are the ones that @kt10aan reported in an old email of ours, and [these](https://gist.github.com/gsaponaro/79b2fef49c9ecdde1543c899e98274dd) are the ones that @jgvictores posted a few messages above. The user should have access to getter and setter methods for RGB modes as well as for Depth modes, and it should be forbidden to try to set an impossible one, for example "640x480@60fps". The way that @kt10aan's driver handles this is pretty simple and effective: the user just calls the driver with the ```--depthVideoMode <num>``` option, where ```<num>``` is the integer corresponding to one of the allowed RGB or Depth modes (listed by ```--printVideoModes```). Looking at the gists that I posted, note that ```<num>``` (obtained from OpenNI2 ```getSupportedModes```) can have a different meaning from sensor to sensor, for example Depth mode 6 sometimes refers to "320x240@25", other times to "160x120@30". There are pros and cons in letting the user select this mode integer versus the actual resolution/FPS explicitly.

I hope I didn't repeat the obvious, or make things more confusing. :)

barbalberto, 2016-11-03:
To avoid compatibility problem, a new device is being written complying with new interface defined in [IRGBD_interface_v2 branch](https://github.com/robotology/yarp/blob/IRGBD_interface_v2/src/libYARP_dev/include/yarp/dev/IRGBDSensor.h) as stated in issue #643 
This device will have only rgb + depth functionality, no skeleton.

drdanz, 2016-11-28:
@barbalberto #974 was merged. Can we close this now?

barbalberto, 2016-11-28:
Well... migration is yet to be done, but maybe it's better to close this one and open a new one later on, starting from what we have now.

closed 2016-11-28 by barbalberto