robotology/yarp #691: Port open() method crashes in Java bindings - Windows

kt10aan, 2016-02-24:
A software we have in Java that is using JYarp bindings is crashing when it is opening a port. We also tried to run the Example1.java, and it also crashes with the same error.

The platform is Windows 7. The version of the compiler used is MSVC12. Everything (windows/yarp deps/Java SE and JDK) is 64-bit.

When the open() method is called, the port does appear in the yarp server but instantly the program crashes before proceeding to the next line of code.

https://gist.github.com/kt10aan/6f45981db56d28af70f0

I haven't tried yet to run the Example1.java in Linux. And there is a good chance that something is my fault, as I am not often compiling YARP and the bindings on Windows (everything was working fine though in another windows machine 3-4 months ago).

Also, it was a clean install, so there are no leftovers from previous installations of yarp and bindings. 

I tried to use the libjyarp.so by compiling the bindings from $YARP_ROOT and by compiling the bindings from $YARP_ROOT/bindings.

Any ideas of what may have gone wrong?

drdanz, 2016-02-25:
I'm not sure if this is useful, but I can confirm that java bindings work on linux, tested right now using `Example1.java`.

kt10aan, 2016-02-25:
@drdanz, thanks! It is useful, as that means that the issue is Windows-specific. Is there any possibility that the issue is caused by the 64-bit bit chain? In my previous windows installation, everything was 32-bit.

drdanz, 2016-02-25:
@kt10aan: @fstramandinoli and I managed to get it running at this year summerschool with 64 bit yarp and matlab, therefore either it is some recent change, or it is something broken in your setup.

kt10aan, 2016-02-25:
Just to be thorough:

- Is there still a difference for bindings (I am referring to all bindings in general) if I compile them from ``$YARP_ROOT`` or from ``$YARP_ROOT/bindings``?

- For Java bindings in Windows, what is the procedure for properly installing them? Apart from making sure that the libjyarp dll is in the java library path (e.g., in my eclipse project), do I have to setup anything additionally? The instructions on the wiki are a bit vague, compared to other bindings.

Since I rarely use Windows (and I also rarely use Java!), I cannot remember the steps I followed last time... I am just trying to figure out if the mistake is on my side (most likely) or YARP/swig's side.

drdanz, 2016-02-25:
* You should compile them from `YARP_ROOT`, the other method is deprecated and it is going to disappear (we already have a branch to remove it), therefore I strongly recommend not to use it.

* The most likely reason why the procedure on the wiki is vague is that the java bindings were not tested deeply by anyone. Whatever you can find out about this is welcome, I'd like (one day) to be able to "make install" and get all the bindings in the right place. If you tell me where the files should be, I'll be glad to fix the installation and the build system.

kt10aan, 2016-02-29:
The issue was solved by manual selecting the toolchain in the initial Configure stage of CMake. Somehow, it was defaulting to 32-bit before.

Regarding the java installation prefixes, I am not the most suitable person to comment on that. I use the python bindings 99% of the time.
Thank you for your answers :) Closing the issue.

closed 2016-02-29 by kt10aan