robotology/yarp #692: [RateThread] getEstPeriod vs. getEstUsed

rlober, 2016-02-25:
Hey guys, this should be a lay up.

Can someone explain the difference between `getEstPeriod(double &av, double &std)` and `getEstUsed(double &av, double &std)`?

I understand that `getEstUsed(double &av, double &std)` returns the time it takes to execute `run()`
(per API):

> Return estimated duration of the run() function since last reset. 

... but I don't understand what `getEstPeriod`'s 

> Return estimated period since last reset.

indicates?

Thanks in advance!

barbalberto, 2016-02-26:
Hi @rlober 

The ```getEstPeriod``` gives the time between two subsequent calls of the ```run()``` function (mean over a number of cycles).
The RateThread is a periodic thread supposed to run with a certain period. This function give you the real period the thread is running at, so you can verify if the request is satisfied.

rlober, 2016-02-26:
@barbalberto Thank you very much!

closed 2016-02-26 by rlober