robotology/yarp #693: Create a yarp module template using yarp companion

apaikan, 2016-02-26:
Up to now yarp companion (e.g., `yarp cmake`) allows to create only a yarp cmake template.  
It would be useful to create a complete yarp module template (i.e., rfmodule, module xml file, cmake with installation,...) using yarp companion. 

suggestions: 
- `$ yarp  create module` 
- `$ yarp module`
- ?

apaikan, 2016-02-26:
more suggestions: 
 - `yarp create project MyModule`
 - `yarp create project MyModule --with-idl`

 we can also move it to the yarpmanager-console: 
- `yarpmanager-console --create-project MyModule`
- `yarpmanager-console --create-project MyModule --with-idl`

Later this can become a graphical wizard in the `yarpbuilder`!

pattacini, 2016-02-26:
If we go for the `yarp create project ...` solution, I'd say we could more suitably transform the command `yarp cmake` into a `yarp create cmake`.

drdanz, 2016-02-26:
I like `yarp project`
* `yarp project module` (create a module skeleton)
* `yarp project module-with-idl` (create a module skeleton with idl support)
* `yarp project cmake` (`yarp cmake`)
* `yarp project device` (create a device skeleton)
* `yarp project carrier` (create a carrier skeleton)
* `yarp project portmonitor-dll` (create a portmonitor dll skeleton)
* `yarp project portmonitor-lua` (create a portmonitor lua skeleton)
and so on.

Perhaps we should have some templates that are not hardcoded in yarp, and that can be extended by adding files in the right folder

barbalberto, 2016-02-29:
If this is meant only for creation, I guess

* `yarp create module` (create a module skeleton)
* `yarp create module-with-idl` (create a module skeleton with idl support)
* `yarp create cmake` (`yarp cmake`)
* `yarp create device` (create a device skeleton)
* `yarp create carrier` (create a carrier skeleton)

is more clear that this command is creating something.

If we think to have also something like:
* `yarp project list`
* `yarp project install [MyProject]`
* `yarp project configure [MyProject] for robot [robotName]`   (just shooting in the dark, but could be nice)

Then for creating new stuff the command could be:
* `yarp project create module` (create a module skeleton)
* `yarp project create module-with-idl` (create a module skeleton with idl support)
* `yarp project create cmake` (`yarp cmake`)

A bit more verbose but more understandable. The more the command is self-explanatory, the less documentation needs to be verbose.

bhigy, 2016-12-01:
Sounds like a nice feature. If we decide how we want to do, I can work on it.