robotology/yarp #701: Intermmitent failure of OS::ThreadTest in Travis

traversaro, 2016-03-02:
* https://travis-ci.org/robotology/yarp/jobs/112846491
* https://travis-ci.org/robotology/yarp/jobs/113122764 
* https://travis-ci.org/robotology/yarp/jobs/111962147

All the failed build have operating system OS X , compiler clang and 
`TRAVIS_WITH_ACE=true TRAVIS_WITH_CXX11=false TRAVIS_WITH_INTEGRATION_TESTS=true TRAVIS_CMAKE_GENERATOR="Unix Makefiles" TRAVIS_BUILD_TYPE=Release` options. 

Error: 
~~~
45: yarp: 1 | ThreadTest:   FAILURE in file /Users/travis/build/robotology/yarp/tests/libYARP_OS/ThreadTest.cpp:424 [IDs match appropriately] t[i].staticId (225460224) == t[i].dynamicId (224919568)
~~~

drdanz, 2016-03-02:
This looks very similar to the race condition that I fixed in 2d0116670feff0a780ba88c818bb8ea916c7ece3, that was supposed to be only for std::thread and pthread, but apparently happens on OSX with ACE as well. The internal implementation is probably different.
It would be useful if someone with OSX could test it locally. If this is reproducible, removing the `#if defined(YARP_HAS_CXX11) || !defined(YARP_HAS_ACE)` at line 91 of [`src/libYARP_OS/src/ThreadImpl.cpp`](https://github.com/robotology/yarp/blob/master/src/libYARP_OS/src/ThreadImpl.cpp#L91) and the following `#endif` at line 93 might fix the issue. Can anyone try this?
Otherwise I can try testing it on travis, but it's going to take a lot more time...

traversaro, 2016-03-03:
I can check it.

traversaro, 2016-03-04:
Reproduced with 
~~~
ctest -R OS::ThreadTest --repeat-until-fail 1000 --output-on-failure
~~~
for two times, once after ~350 runs and another after ~750 runs. Testing now the fix.

traversaro, 2016-03-04:
With the fix is running fine for ~900 runs, opened https://github.com/robotology/yarp/pull/703 .

PR #703 (2016-03-09): YARP_OS: fix race condition in setting thread id
Fix #701
closed 2016-03-09 by drdanz