robotology/yarp #704: Fix version badge and link to releases instead of tags

Tobias-Fischer, 2016-03-09:
Hi all,

I noticed that the wrong version number of Yarp is shown on https://github.com/robotology/yarp

Also, the link guides to `tags` rather than `releases`, which I think is more correct.

Both can be fixed as follows:
```
diff --git a/README.md b/README.md
index b132ad4..98d8a12 100644
--- a/README.md
+++ b/README.md
@@ -8,7 +8,7 @@ YARP
 ====
 
 [![YARP Homepage](https://img.shields.io/badge/YARP-Yet_Another_Robot_Platform-orange.svg)](http://wiki.icub.org/yarpdoc/)
-[![Latest Release](https://img.shields.io/github/tag/robotology/yarp.svg?label=Latest Release)](https://github.com/robotology/yarp/tags)
+[![Latest Release](https://img.shields.io/github/release/robotology/yarp.svg?label=Latest Release)](https://github.com/robotology/yarp/releases)
 
 YARP is a library and toolkit for communication and device interfaces,
 used on everything from humanoids to embedded devices.
```

Best,
Tobi

drdanz, 2016-03-09:
@Tobias-Fischer Thanks.
This seems to be the reason for this: https://github.com/badges/shields/issues/490
We have a tag `start` that probably confuses shields.io.

closed 2016-03-09 by drdanz

drdanz, 2016-04-06:
FYI, badges/shields#490 is now fixes and  [image] should report the latest tag

commit 95bdb06f (2016-03-09): README.md: Fix broken tags badge

Fixes #704

Thanks @Tobias-Fischer