robotology/yarp #709: Windows IoT, C# and YARP

roby85, 2016-03-15:
I developed several C# modules that communicate with each other through YARP. 
For this purpose, YARP pre-compiled version 2.3.63 for Windows has been used.

I have tried unsuccessfully to use some of these modules on Raspberry Pi2 with Windows IoT. 

Is there any YARP pre-compiled version for Windows IoT? 
If not, how can I compile YARP for Windows IoT? 
thanks

drdanz, 2016-03-16:
I made a few tests some time ago, but unfortunately YARP packages for Visual Studio 2015 (required by Windows IoT) are not available yet (I think that ACE had some issue).

I never tried it for Windows IoT, but you can try compiling YARP without ACE (it will have limited functionalities) by enabling the `SKIP_ACE` flag in CMake.
You will need CMake 3.4 or later and Visual Studio 2015.

Instructions for building YARP on Windows are [here](http://www.yarp.it/install_yarp_windows.html) even though you might have to change something for the IoT version.

Let us know if you manage to get it working or if you are having problem!

drdanz, 2016-04-13:
Closing since no feedback was received in one month.

closed 2016-04-13 by drdanz