robotology/yarp #711: Python binding for ResourceFinder::configure

clement-moulin-frier, 2016-03-16:
Hi,

I am trying to use the `RFModule` class with the Python bindings. It works well but I can't figure out how to configure the `ResourceFinder`.

Here is the code I use:

```
import yarp
import time

import sys

class Module(yarp.RFModule):
    def configure(self, rf):
        self.out_port = yarp.Port()
        self.out_port.open("/test/out:o")
        return True
    def close(self):
        print "Closing ports"
        self.out_port.close()
    def interruptModule(self):
        return True
    def getPeriod(self):
        return 1.
    def updateModule(self):
        b = yarp.Bottle()
        b.addString("aaa")
        print b.toString()
        self.out_port.write(b)
        return True

yarp.Network.init()
mod = Module()
rf = yarp.ResourceFinder()
rf.setVerbose(True);
rf.setDefaultContext("myContext");
rf.setDefaultConfigFile("default.ini");

# The following line causes trouble
rf.configure(len(sys.argv),  sys.argv)

mod.configure(rf)

mod.runModule(rf)
```

When I execute it, I receive the following error at the line `rf.configure(len(sys.argv),  sys.argv)`:

```
||| clearing context
||| adding context [sensationManager]
Traceback (most recent call last):
  File "yarp_python.py", line 50, in <module>
    rf.configure(len(sys.argv),  sys.argv)
  File "/home/clement/dev/yarp/build/lib/python/yarp.py", line 3966, in configure
    return _yarp.ResourceFinder_configure(self, *args)
NotImplementedError: Wrong number or type of arguments for overloaded function 'ResourceFinder_configure'.
  Possible C/C++ prototypes are:
    yarp::os::ResourceFinder::configure(char const *,int,char *[],bool)
    yarp::os::ResourceFinder::configure(char const *,int,char *[])
    yarp::os::ResourceFinder::configure(int,char *[])
    yarp::os::ResourceFinder::configure(std::string const &,std::vector< std::string,std::allocator< std::string > > &,bool)
    yarp::os::ResourceFinder::configure(std::string const &,std::vector< std::string,std::allocator< std::string > > &)
```

In C++, the call to `configure`  normally is `rf.configure( argc, argv);`, where `argc` and `argv` are provided by the arguments of the `main` function. However it is not clear how to use it in Python. From the error message above, we see that a possible prototype is:
```
yarp::os::ResourceFinder::configure(std::string const &,std::vector< std::string,std::allocator< std::string > > &)
```

However by replacing the problematic line with `rf.configure("test", ["test", "test"])`, the same error occurs.

I've also tried:
```
from ctypes import *
argv = ["test1", "test2", "test3"]
rf.configure(len(argv), (c_char_p*len(argv))(*argv))
```

with still the same error.

Any idea about how to deal with this?

Thanks a lot,
Clément

clement-moulin-frier, 2016-03-16:
Update: it works when replacing the problematic line by:
```
rf.configure(0, None)
```

I can then access to the `.ini` file :)

However, it prevents to use useful command line arguments like `--from other_config.ini`

traversaro, 2016-03-16:
Some years ago [1] the good old @elen4 add some nice helpers to support the `rf.configure` out of the box in Lua, to write Lua script behaving like a YARP Module. You can see an example of its use in the `IOL` Lua helpers [2]. 

Apparently that uses a SWIG macro contained in the `swiglib/lua/argcargv.i` file. Considering that a similar file is contained in upstream SWIG since a long time (see https://github.com/swig/swig/blob/rel-2.0.3/Lib/python/argcargv.i) probably you can add some lines to the `yarp.i` to obtain something similar to what @elen4 obtained for Lua. 

[1] : https://github.com/robotology/yarp/commit/7e7a57488b33ff5d8f01e3de459a8d9cfbc98383
[2] : https://github.com/robotology/iol/blob/master/app/lua/iol_main.lua#L46
[3] : https://github.com/robotology/yarp/blob/master/bindings/yarp.i#L29

traversaro, 2016-03-16:
See also:
http://swig.10945.n7.nabble.com/wrapping-main-int-argc-char-argv-td8009.html
http://www.swig.org/Doc2.0/Typemaps.html#Typemaps_multi_argument_typemaps

elen4, 2016-03-17:
@traversaro what do you mean, "old"? :angry: 
Anyway yes, it should only be a matter of adding 

```
#if defined (SWIGPYTHON)
  %include "argcargv.i"
  %apply (int ARGC, char *ARGV[]) { (int argc, char *argv[]) }
#endif
```

to `yarp.i` and regenerate the bindings to have it working, but I cannot test it right now :neutral_face:

closed 2016-03-18 by drdanz

drdanz, 2016-03-18:
@clement-moulin-frier Tested, it should be working now...

@elen4 Thanks a lot for the hint! Anyway I had to change

```diff
-%apply (int ARGC, char *ARGV[]) { (int argc, char *argv[]) }
+%apply (int ARGC, char **ARGV) { (int argc, char *argv[]) }
```
because the `argcargv.i` for python and ruby were slightly different from the one for lua. I also changed the lua version to do the same as the others...

clement-moulin-frier, 2016-03-18:
Many thanks guys, just tried it and it works perfectly (I can specify the config file using `--from`).
Feel free to include this script in the `examples` directory if you want:

```

import yarp
import time
import sys

print sys.argv

class Module(yarp.RFModule):
    def configure(self, rf):
        self.out_port = yarp.Port()
        self.out_port.open("/test/out:o")
        moduleName = rf.check("name", yarp.Value("SensationManager")).asString()
        self.setName(moduleName)
        print moduleName

        grp = rf.findGroup("SENSATIONS");
        self.sensationList = grp.find("sensations").asList();  
        for i in range(self.sensationList.size()):
            print mod.sensationList.get(i).asString()        
        return True
    def close(self):
        print "Closing ports"
        self.out_port.close()
    def interruptModule(self):
        self.out_port.interrupt()
        return True
    def getPeriod(self):
        return 1.
    def updateModule(self):
        b = yarp.Bottle()
        b.addString("ici")
        print b.toString()
        self.out_port.write(b)
        return True
    

yarp.Network.init()

mod = Module()

rf = yarp.ResourceFinder()
rf.setVerbose(True);
rf.setDefaultContext("myContext");
rf.setDefaultConfigFile("default.ini");
rf.configure(sys.argv)

mod.configure(rf)

mod.runModule(rf)

```

commit 85d5a5ff (2016-03-18): Bindings: Handle argc and argv in python and ruby

Fixes #711