robotology/yarp #715: Long delay on closing a connected port (similarly on disconnect)

apaikan, 2016-03-30:
The issue was initially reported by @pattacini  ! 

There is a long delay on closing a yarp `Port` / `BufferedPort` :

```c++
int main() {
    Network yarp;
    Port port1;

    double t0=Time::now();
    port1.open("...");
    double t1=Time::now();

    NetworkBase::connect(port1.getName(), "/r", "udp");

    double t2=Time::now();
    port1.close();
    double t3=Time::now();

    yInfo()<<"opening "<<1000.0*(t1-t0)<<" [ms]";
    yInfo()<<"closing "<<1000.0*(t3-t2)<<" [ms]";

    return 0;
}
``` 

Output: 
```
yarp: Removing output from /tmp/port/1 to /r
PortCore::removeUnit(): 252.91
[INFO]opening  7.061  [ms] 
[INFO]closing  256.231  [ms] 
```

A quick check has shown that the source of the issue probably is in `PortCore.cpp` within `PortCore::removeUnit()` while waiting on `stateMutex`:

```c++
bool PortCore::removeUnit(const Route& route, bool synch, bool *except) {
...
            if (synch) {
                // Wait for connections to be cleaned up.
                YARP_DEBUG(log,"synchronizing with connection death");
                bool cont = false;
                do {

                    double t1 = yarp::os::Time::now();
                    stateMutex.wait();                    
                    for (int i=0; i<(int)removals.size(); i++) {
                        cont = isUnit(route,removals[i]);
                        if (cont) break;
                    }
                    if (cont) {
                        connectionListeners++;
                    }                    
                    stateMutex.post();
                    double t2 = yarp::os::Time::now();
                    printf("PortCore::removeUnit(): %.2f\n", (t2-t1)*1000.0);

                    if (cont) {
                        connectionChange.wait();
                    }
                } while (cont);
            }
...
```

drdanz, 2016-04-13:
I tested this some days ago, but unfortunately I wasn't able to reproduce the delays. Opening and closing times are around the same (<10 ms).
Perhaps there is some extra variable to take into account.

Have you tried to reproduce it in a completely clean setup?

pattacini, 2016-04-13:
@drdanz the macroscopic effect we noticed on the robot is that the module `ARE`, for example, takes very long to close all its motor boards, while it's pretty fast at start up. This observation triggered our attention and motivated further investigations.

To confirm our intuitions I then came up with the first snippet above, that however didn't show any particular difference between opening and closing on my Windows machine, while @apaikan caught again the problem on its Linux and reported here.

I don't know what is the current status by now, but I'm sure that the observations on `ARE` are deterministic (e.g. on Blacky's system).