robotology/yarp #718: problems running yarp server on a machine with multiple network interfaces

randaz81, 2016-03-31:
Example:
Machine A, on which yarp server runs, has two network interfaces:
* 192.168.100.10 (this ip address is visible to all machines belonging to the 192.168.100.x net)
* 10.0.1.104 (this ip address is the master address for all ETH motor control boards).
There is no bridge between the network interfaces. 

Another machine B, with ip address 192.168.100.20, is configured to find yarp on machine A (`yarp conf 192.168.100.10:1000`).
On B, Yarp is successfully detected and command `yarp name list` shows all the available yarp ports.
HOWEVER, the ip address binded by yarp for these ports is 10.0.1.104, so even if yarp port names are visible from B, any attempt to read a port (using `yarp read`, for example) fails. Instead, if the same yarp read command is executed on A, the command is successful.

I remember that this issues exists since many many years. 
The typical workaround is to configure yarp server on machine A to bind the correct address (`yarp conf 192.168.100.100.10:10000`), remembering that if yarp server is then moved to another machine, this configuration is lost and that just running `yarp server` will probably bind again to the wrong ip address 10.0.1.104.

I think that the fact that `yarp name list` command successfully display ports which cannot be later accessed (unless you are on that specific machine) is misleading trap: users typically waste a lot of time before realizing which is the origin of the problem, usually thinking to a firewall problem.

lornat75, 2016-03-31:
This behavior is intended and not a bug (I agree it may sound confusing). I think it allows clients to register and get port registrations across different network. This is rarely used by us, but it seems like a reasonable use case. Maybe we can work on how to reduce confusion.

barbalberto, 2016-03-31:
There was a long discussion about this 2 years ago in issue #142 and #143, just to recap ideas

randaz81, 2016-04-01:
One possibility I think could add a little more clarity is the following.
We can add a file `blacklist.conf` where a list of ip addresses is written.
When `yarp server` is started, if the ip address automatically chosen by the server (or chosen by the user in yarp.conf) matches one of the blacklisted addresses, an error message is displayed and yarp server terminates (the implementation should be pretty easy and straightforward)
In this way the user is always notified if yarp server is accidentally started on the wrong network interface.

drdanz, 2017-02-28:
Closing this and reopening the original issue #143

closed 2017-02-28 by drdanz