robotology/yarp #722: yarpserver used with --ros should not start if required carriers are not found/compiled

barbalberto, 2016-04-01:
yarpserver used with --ros option is meant to work together with roscore and therefore it requires the following carriers:

- rossrv
- tcpros
- xmlrpc

right now the server does not check if the plugins are available, and at the connection time it prints an error

yarp: Failed to find library support for carrier xmlrpc
yarp: Could not find carrier "xmlrpc"

which is kind of difficult to spot. 
IMHO it'd be better if the server does not start if the carriers are not available in order to catch the problem as soon as possible.

drdanz, 2016-04-20:
Does it work with xmlrpc, but not rossrv and tcpros?

aerydna, 2016-07-04:
also, if you start `yarpserver --ros` without such plugins and then you call `yarp plugin *anything*` yarpserver crash (segfault).

PR #1679 (2018-05-10): Add check on carriers for yarpserver --ros
It fixes #722 .

This PR makes `xmlrpc` `rossrv` `tcpros` carriers mandatory for running `yarpserver --ros`.

Please review code.
commit ceadca91 (2018-05-10): yarpserver: abort if `rossrv` or `tcpros` or `xmlrpc` are not found when
launched with `--ros` option

It fixes #722
closed 2018-05-10 by drdanz