robotology/yarp #725: yarpmanager++ compile on windows

apaikan, 2016-04-07:
`yarpmanager++` contains the new version of the yarpmanager and builder based on QT. The project compile correctly on Linux machine but failed to compile on Windows: 

```
error C2491: 'BuilderWindow::staticMetaObject' : definition of dllimport static data member not allowed	f:\robotology\yarp\build\src\yarpmanager++\moc_builderwindow.cpp
```

QT version: Qt-5.4.1_v10_x86
Visual Studio: 10 
Windows: 7, X86 

CC: @aitek4iit  
CC: @drdanz

pattacini, 2016-04-07:
:cake: for the new yarp baby :tada:

apaikan, 2016-04-08:
@aitek4iit  

I have tried both debug and release mode. The Cmake file is same as the previous one (yarpmanager-qt).  

- Should the builder source files compiled as a separate library? I am compiling manager and builder as a single project. Anyway this compile fine on Linux. 

- Does the builder requires specific package from Qt? It can be that something missing in our installation? I am compiling as before using Widget and Gui. 
```
qtyarp_use_modules(yarpmanager++ Widgets Gui)
```

- How the builderwindow differ from other windows? static class / member function, etc?

 - Did you tried to compile using Visual studio too? 

The error is in the moc_builderwindow.cpp:
```
const QMetaObject BuilderWindow::staticMetaObject = {
    { &QWidget::staticMetaObject, qt_meta_stringdata_BuilderWindow.data,
      qt_meta_data_BuilderWindow,  qt_static_metacall, Q_NULLPTR, Q_NULLPTR}
};
```

aitek4iit, 2016-04-08:
I think you have to define YARPBUILDERLIB_LIBRARY somewhere.
If you take a look at the .pro file of the qt creator project you will see the qt dependencies (widgets opengl) and the defines. 
I always compiled it as a dynamic library.

drdanz, 2016-04-08:
@apaikan I'm not sure if we are going to have a separate library for the builder, but just removing  `YARPBUILDERLIB_LIBRARY` from [builderwindow.h:30](https://github.com/robotology/yarp/blob/yarpmanager%2B%2B/src/yarpmanager%2B%2B/src-builder/builderwindow.h#L30) fixes the build...

apaikan, 2016-04-11:
Thanks @aitek4iit  and @drdanz. It works! 

All the builder codes are kept in a separate folder (i.e., `yarpmanagewr++/src-builder`) and creating a separate library for builder should be easy. 

However, for now we compile both builder and manager together to avoid having an extra library in YARP (for builder) which has not been used in any other projects within YARP.

closed 2016-04-11 by apaikan